RavenSystem/esp-homekit-devices #1830: Initial State Problem.

TheReiner, 2022-05-21:
Hi,

This is 2 configurations

{"c":{"l":13,"b":[{"g":0,"t":5}],"u":1,"s":5,"n":"guirlande_ext"},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]}]}

{"c":{"l":13,"b":[{"g":0,"t":5}],"u":1,"n":"guirlande_ext"},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]}]}

If i use the first configuratio with Initial State ("s":5), the device not appear on category "HAA" on  HAA Manager on my iPhone ... It's on "Others" device not "HAA", and i can't use Update, Setup, Reboot or Wifi Reconnection

Any suggestion ?
Thanks

steini72, 2022-05-21:
Initial State ("s":5), must be under "a":
"a":[{"s": 5,"0":{"r":[{"g":12}]},...............

{"c":{"l":13,"b":[{"g":0,"t":5}],"u":1,"n":"guirlande_ext"},"a":[{"s": 5,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]}]}

TheReiner, 2022-05-21:
Many Thanks ;) It's work

closed 2022-05-21 by TheReiner