RavenSystem/esp-homekit-devices #1835: Dimmable lamps blink, do not switch off as Shelly Dimmer 2 goes off the rails

tokamac, 2022-05-27:
**Used device**
Shelly Dimmer 2

**HAA version**
10.4.0 to 10.6

**iOS/iPadOS version**
iOS 15.5

**JSON String**
`{"c":{"r":[{"n":0}],"l":16,"i":0,"b":[{"g":13,"t":5}]},"a":[{"t":30,"b":[{"g":13},{"g":12,"i":1,"p":0,"t":0},{"g":14,"i":1,"p":0},{"g":14,"t":0}],"0":{"u":[{"v":"010202060000f401000000ff04"}]},"y0":[{"v":1,"0":{"u":[{"v":"010202060a00f4010000010904"}]}},{"v":5,"0":{"u":[{"v":"010202063200f4010000013104"}]}},{"v":10,"0":{"u":[{"v":"010202066400f4010000016304"}]}},{"v":15,"0":{"u":[{"v":"010202069600f4010000019504"}]}},{"v":20,"0":{"u":[{"v":"01020206c800f401000001c704"}]}},{"v":25,"0":{"u":[{"v":"01020206fa00f401000001f904"}]}},{"v":30,"0":{"u":[{"v":"010202062c01f4010000012c04"}]}},{"v":35,"0":{"u":[{"v":"010202065e01f4010000015e04"}]}},{"v":40,"0":{"u":[{"v":"010202069001f4010000019004"}]}},{"v":45,"0":{"u":[{"v":"01020206c201f401000001c204"}]}},{"v":50,"0":{"u":[{"v":"01020206f401f401000001f404"}]}},{"v":55,"0":{"u":[{"v":"010202062602f4010000012704"}]}},{"v":60,"0":{"u":[{"v":"010202065802f4010000018b04"}]}},{"v":65,"0":{"u":[{"v":"010202068a02f4010000010904"}]}},{"v":70,"0":{"u":[{"v":"01020206bc02f401000001bd04"}]}},{"v":75,"0":{"u":[{"v":"01020206ee02f401000001ef04"}]}},{"v":80,"0":{"u":[{"v":"010202060032f4010000013104"}]}},{"v":85,"0":{"u":[{"v":"010202065203f4010000015404"}]}},{"v":90,"0":{"u":[{"v":"010202068403f4010000018604"}]}},{"v":95,"0":{"u":[{"v":"01020206b603f401000001b804"}]}},{"v":100,"0":{"u":[{"v":"01020206e803f401000001ea04"}]}}]}]}
`

**Detailed steps to reproduce it**
I have a serious issue with [Shelly Dimmer 2](https://shelly.cloud/products/shelly-dimmer-2-smart-home-light-controller/) and HAA whatever the version between 10.4.0 and 10.6 (I downgraded to 10.4.0 from 10.6 as at first I thought I was reproducing the issue described in #1806, but that's not the case).

Here are all the details.

The goal is to switch on/off and dim a ceiling light made of three same 230V Osram Parathom Superstar PAR16 GU10 LED dimmable glass reflector lamps, 4.5W each:

[image]

Total power = 13.5W, therefore as it is over 10W the Shelly bypass is not mandatory. 
The bulbs are brand new, bought on Amazon last week (ASIN B09F9F1BQD) and their cardboard box show the dimmable logo indeed. 
There is no neutral wire in the electrical wall switch box.
For now I have a one-way switch at the wall (was planning to replace it with a momentary push button if all worked as intended).

I bought two brand new Shelly Dimmers 2. I connected the wires exactly as told by Shelly, and tested both dimmers with the original Shelly firmware. **All is working perfectly**: the three bulbs of the lamp switch on and off together, and they dim properly. Either from the wall switch or from the Shelly web app.

Then, I upgraded to HAA. I tested the 10.5.x branch, current latest 10.6 version, then even downgraded to 10.4.0: Nothing works properly, on both dimmers. Here is the chronological process:

1. I switch off the differential circuit breaker to install the module into the wall box, with the live wire connected to the second "L" connector and the switched live of the lamp to "O", like this: 

[image]

2. I test first with Shelly FW, everything is fine. lamp has an IP on the local network now. I upgrade the FW to HAA through OTA using tasmota-minimal as a temporary solution, as explained in the wiki. HAA installs fine. 
3. I use the JSON provided in the wiki for that specific device, the one pasted above.
4. Inclusion in HomeKit works fine.
5. **I switch on the light: all bulbs blink rapidly like a stroboscope!** The device is still operational in Apple's Home app.
6. Then I switch off the light, or change its brightness level (either with the switch, or with HomeKit): **the lamp continues to flash rapidly, it doesn't switches off! And at this very moment, when status is changed to off or dimm, the device is no more reachable in Home or HAA manager**, like if it was completely shut down. The lamp continues to blink until I cut the differential circuit breaker off in the panelboard. 
7. Once I have put the differential on again, the lamp remains in the off state and the device shows up in HomeKit and HAA manager again as if there was no problem. But it systematically fails as soon as point (5) is triggered, no matter how.

I tried with the wall switch connected to L on one hand and SW1 or SW2 on the other, but the behavior is the same. Tried simple or double click: same issue. Exact same thing with no switch connected at all to the Shelly Dimmer 2, operated in this case with two wires (L and O) with HomeKit only.

What happens?

tokamac, 2022-05-31:
Sorry, I don't understand what "laying a temporary zero" means. 

As for testing with another lamp, I just moved to a new place and it happens that every lighting here is LED GU10 and I have no access to any "more standard" and more powerful lamp. Sure there is this old G9 halogen ceiling lamp in the bathroom but I have no desire to dim it and whatever, I will get rid of it as soon as possible and replace it with a less power hungry LED.

My guess for now is that 13.5W is a lower threshold and it goes wrong because of this too low power. I ordered a Shelly bypass to test this hypothesis. Waiting for it.

This, or these specific bulbs, although advertised as "dimmable", are not compatible with HAA. But really, I don't understand why there is no issue with the stock Shelly firmware.

steini72, 2022-05-31:
[image]
I meant a temporary neutral conductor.

tokamac, 2022-05-31:
Oh, I see! I didn't test with a neutral wire but I think I could, wiring up the Shelly dimmer not into the switch box, but directly onto the ceiling lamp wiring. Will try tomorrow!

steini72, 2022-05-31:
Ok, Please feedback

tokamac, 2022-06-11:
The Shelly Bypass was in the mailbox today, looked like it took an eternity to arrive. So I was finally able to test with the Bypass connected in parallel to the lamp, and the Shelly Dimmer 2 back into the wall box without neutral: and... it worked! Once. 

The first time I connected that setup, all worked perfectly except at 100% brightness, where the bulbs blinked again repeatedly a few times and the lamp eventually shot down. But the dimmer was still available in the Home app, it did not go off the rails as previously without the Bypass. I tried several times and the issue was definitely at 100% brightness only. I could dim under that value, and switch off the light. Good.

At first I thought that the issue was minor and the solution easy: either to limit Max Power, lowering the value with the "mp" variable in JSON; or to change the sequence corresponding to "v":100 to a lower one, since it worked fine with a lower brightness. 

None of them work. Since then, the setup has become unreliable again, exactly like before. I'm at a loss.

closed 2022-07-10 by stale[bot]

mstaack, 2022-11-12:
@tokamac did you ever find a solution?

i am experiencing exactly the same

mstaack, 2022-11-12:
maybe an update to the intern stm32 is needed https://github.com/jamesturton/shelly-dimmer-stm32/releases ?