RavenSystem/esp-homekit-devices #1841: Ubiquity ap’s 6.0.19 update problem

alexmb2020, 2022-06-20:
Hello, 
Upgraded APs (ac pro and wifi6 LR) to latest 6.0.19 version and Raven based sonoff switch with 2 sensors stopped working.
No response in homekit. After reset and removal, I can add this switch to home, it shows active until first click then grey out and no response.
Tried with second new board with final Raven firmware, the same problem.
Downgraded access points to previous version, all works.
Any idea?

rg2k, 2022-07-03:
The latest version is 6.0.21 - I've been running this version with no issues at 2 different sites, and with many Raven based Sonoff's.

alexmb2020, 2022-07-03:
Hello, yes I updated from 6.0.15 yesterday to latest 6.0.21 and All looks good so far. Manually tried 6.0.19 the same problem. So yes its good news 6.0.21 works with Raven boards without problems

Daveblanche, 2022-07-03:
Me too. No problems. Running 2 pro APs in WiFi 6 flavours and 4 others in WiFi 5 without any issues. 
22 sonoff with raven magic in them.  Check that the unifi updates haven’t messed with mDNS, beacon interval or something else  silly. I have had that before now. I do “sometimes” take my life in my hands and use a beta, especially where I see things about leaks, dropped connections.

jlg89, 2022-07-08:
I've been using Ubiquity for several years, and have always had mDNS-related issues with esp-homekit-devices firmware. Only tried it on Sonoff SV boards, but I would imagine it's the same regardless of hardware. In the past I've used avahi on my homebridge server to broadcast a duplicate mDNS record for the Sonoff board. I've just added "ttl" : [ 4500, 300 ] to the "c" section of the JSON config, to see if that will solve it in a more straightforward way.

wsc96, 2022-07-31:
> I've been using Ubiquiti stuff for several years, and have always had mDNS-related issues with esp-homekit-devices firmware, even with AP firmware 6.0.21 (though I only updated a few days ago, so maybe I just haven't given it enough time to settle out). I've only used Sonoff SV boards, but I would imagine it's the same regardless of specific hardware. In the past I've used avahi on my homebridge server to broadcast a duplicate mDNS record for the Sonoff board. I've just added "ttl" : [ 4500, 300 ] to the "c" section of the JSON config, to see if that will solve it in a more straightforward way. Others elsewhere have suggested using "ttl": 60 , but that doesn't match the [documentation](https://github.com/RavenSystem/esp-homekit-devices/wiki/General-Configuration#HomeKit-mDNS-TTL).
> 
> Update: after several days running with the ttl entry, I can say with certainty that it helps. Whereas the SV used to run maybe 50/50 in terms of HomeKit availability, it's now probably >95%. I may need to tweak that 300 a bit lower, but in general I would say that there should probably be something in the documentation that recommends the ttl tweak for anyone using Ubiquiti WiFi.

Hi @jlg89 ,

Still having success with the specified TTL values? I'm getting a few random HAA devices reporting no response yet still on the network just fine and might try this too.

Regards,

jlg89, 2022-08-02:
So far so good. Every now & then the board shows "not responding" in the Home app, but it's very infrequent, and if it ever bothers me, I reckon I can probably fix it by reducing that ttl value a bit more.

rg2k, 2022-08-02:
It may be useful to compare other Wi-Fi settings? I created a separate Wi-Fi network with only 2.4Ghz, and everything basically switched off.  Also, changed 802.11 DTIM Period to 3/3. I've not had a single device go "no response" since I made these changes.

wsc96, 2022-08-02:
> It may be useful to compare other Wi-Fi settings? I created a separate Wi-Fi network with only 2.4Ghz, and everything basically switched off.  Also, changed 802.11 DTIM Period to 3/3. I've not had a single device go "no response" since I made these changes.

Thanks for the suggestion. I'll double check my settings.

wsc96, 2022-08-04:
For anyone that stumbles upon this later; the ttl configuration change per the wiki has made all the difference for me and HAA devices are no longer randomly showing as Not Responding. Back to being rock solid again.