RavenSystem/esp-homekit-devices #1852: Shelly 2.5 lifetime

adamkasprzak, 2022-07-19:
I've got two dead shelly 2.5 devices. Both units failed within a week. What is lifespan of your installation? Mine lasted 2 years. After fresh re-flash I managed to add them to HomeKit, but when I connect 230V, the device is not recognizable, heats up and you may head a hissing sound. The load was insignificant (bedside lamp). Shelly tech support suggested that the unis seem faulty.

I have also Shelly 1 in my "collection" for longer time - I guess those are more durable.

Is there an alternative to Shelly 2.5 (except Sonoff dual R3 which is too bulky and harder to flash)?

pawelosin, 2022-07-23:
Jeśli chodzi o Selly 2.5 to nie mam z nim doświadczenia, używam wielu Shelly 1 i kilka z nich też mi się uszkodziło, natomiast co do modułu jaki ja używam do lampki nocnej to jest Sonoff basic r3. Jest co prawda jednokanałowy, ale wygląda całkiem ładnie i ma wbudowany przycisk, który da się zaprogramować jako włącznik modułu oraz dodatkowo jako przycisk do sterowania innymi urządzeniami.

MagOO33, 2022-08-05:
Same thing with one shelly out of seven. The buzzing sound could be a capacitor issue. Some say that it about replacing the 16V/100uF capacitor.