RavenSystem/esp-homekit-devices #1855: Strange behavior of Garage Door (Probably bug?)

Starksoft, 2022-07-24:
**Detailed steps to reproduce it**
Hi. I'm building garage door remote with Sonoff Mini R2.

I Have Nice Robus 1000 with 433 mHz remote control

STR:
- Open door with 433 mHz remote control
- Stop door with 433 mHz remote control

Current result:
State is wrong
Garage door in Home application indicates Closing... or Opening... endless

Expected result:
State is correct - Opened
After Door Closing/Opening Time state in Home application becomes "Opened"

**JSON String**
{"c":{"l":13,"m":1,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"10":{"a":0},"12":{"a":0},"13":{"a":0},"t":40,"d":5,"vs":1,"f2":[{"g":0,"t":0}],"f3":[{"g":2,"t":0}],"f4":[{"g":2,"t":1}],"f5":[{"g":0,"t":1}],"f8":[{"g":4,"t":0}]}]}

**Used device**
Sonoff Mini R2

**HAA version**
10.7.5

**iOS/iPadOS version**
iOS 15.6

**Logs**
Jul 24 18:38:17 [Test] Ворота тест: Current Door State = 2
Jul 24 18:38:17 [Test] Ворота тест: Target Door State = 0

**Screen-shots or videos showing issue**

Starksoft, 2022-07-24:
I think I figured out what would fix the problem. It is possible to add a timer to reset the state of the garage door to the previous one if there was no successful transition to the target state?

RavenSystem, 2022-07-24:
You are using sensor, so door state is based on sensors. No bug.

closed 2022-07-24 by RavenSystem

Starksoft, 2022-07-24:
There are no sensor of transition state or the door.
Please, read the STR

RavenSystem, 2022-07-24:
[image]

I can see declared sensor for open and close in your JSON.

Starksoft, 2022-07-24:
Yep, the JSON gives us the funtionality:
f2-f5, but if we stop gates from 433 mHz remote, state is always "Opening".

But if I remove "f2" this functionality works perfect, drops the state to previous if desired state is not reached