RavenSystem/esp-homekit-devices #1875: Extra pairing does not work

Starksoft, 2022-08-31:
**Detailed steps to reproduce it**
I followed steps from video https://www.youtube.com/watch?v=rtkBfCiT0MY&t=4s&ab_channel=RavenSystem
And nothing happens.
Also in setup mode i have 4/48 paired devices, but I have only one :(
I erased the firmware completely and installed a new one via cable - no effect

**MEPLHAA Script**
Any, for example:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]}]}

**Used device**
SOnoff mini r2
**HAA version**
11.0

**iOS/iPadOS version**
15.6.1

**Logs**
A .txt file with logs. Read [how to capture logs from your device](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs)
Logs is empty. Capturing throw HAA Manager

**Screen-shots or videos showing issue**

RavenSystem, 2022-08-31:
If you have 4 pairings, it is because there are 4 users added. Most common case is that you have 3 guests in your home.

When you say "And nothing happens." what means, because in the video is working fine.

Starksoft, 2022-08-31:
Thanks, 4 users at home - ok

Nothing happens - means no extra pairing is occurred.
Video: https://photos.app.goo.gl/gMZ6HSe8chk6mMuk7

VaBanck, 2022-08-31:
https://discord.gg/7VKeQGr9 заходи сюда

RavenSystem, 2022-08-31:
Are you sure that your device is not currently paired with the Sonoff?

Starksoft, 2022-08-31:
Yep, thats the problem. 
I was trying to pair with home from same user.
Can You add feature to add extra users to shared home from same user?

Scenario:
1. Full home on user@appleid.com
2. Shared home on the same user user@appleid.com

This would allow more flexibility in setting up sharing between users, keeping access control

closed 2022-08-31 by RavenSystem