RavenSystem/esp-homekit-devices #1882: Roller shutter with remote control

freedreamer82, 2022-09-13:
Hi,

first of all , very nice proect! I already bought the ios app and it works pretty well.
I already set up some devices in my home with sonoff but  I would like to integrate the esp-homkit-devices in my  roller shutter.
I have a system with a remote where you can select trough a button  the number of the shutter and you have 3 other buttons as well to stop/open/close. (very common interface).
I already set up the script to show the n shutters on the ESP but now i have to connect the remote buttons to the controller.
The idea is to solder wires to the buttons and connect them to a ESP board.
Unfortunately the selection button has a sort of memory...so if i press twice the next time it will start from here...how can I handle this behaviour in the script section ?
Moreover..how do u suggest to implement the consecutive pressures ? maybe with a pwm ?

thanks for your tips.

Regards

              Marco

freedreamer82, 2022-09-24:
no tips ? ;(

M.

ceg117, 2022-10-17:
I have studied blinds or curtains, but it may take 3 relays to jog to achieve the up, down and stop required by you. However, if the opening or closing time of the whole curtain is matched in json, then the opening% can be defined. If the external button is pressed, just jog the starting mcu pin

ceg117, 2022-10-17:
I have studied the shutters or curtains, but it may take 3 relays  jog to achieve the up, down and stop required by you. However, the premise is that the opening or closing time of the whole curtain must be matched in json to define the opening%. If the external button is pressed, just jog the trigger pin

closed 2022-11-23 by stale[bot]