RavenSystem/esp-homekit-devices #1888: Revert to Tasmota GU10 device

csc-solution, 2022-09-28:
Hi,

we have more than 50 GU10 devices and there is no possibility to open it without destroying. It cannot be the point of open source software that after flashing there is a firmware lock and a rollback is no longer possible. Everybody withe in the same situation have to throw all devices in the trash and lose a lot of money.

Please provide an early solution to finally free all devices from the HAA firmware. The easiest is a HAA2Tasmota update, just as easy as Tasmota2HAA. Some other people an i would nbe very thankful for that.

Best regards,
Thomas

RavenSystem, 2022-09-28:
Read IMPORTANT note from: https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation#third-party-installation-tools

closed 2022-09-28 by RavenSystem

csc-solution, 2022-09-28:
I know the notes say "A wired connection is required" but it can't be it. If youre sure there are no other ways then please tell me how should i open some GU10 devices without destroying to flash it with wires?

You said only signed firmware are allowed in case of security reasons, but there are some other ways to do that without a hard lock. For example a special on-off sequence in combine with timeouts, that only people where have the device in his hands can do that.

Think about it that are your devices and you would change to another system... I can tell you, you would be also very angry in this situation.

alexsannikov, 2022-12-11:
I believe it can be a some “limited” version of firmware, which should be definitely signed and locked into “flashing” mode only.
From that version it may be upgrade to other vendor’ FW or back to HAA full if you changed your mind.

panache67, 2024-12-09:
Just wondered if there has been any development in that regard since 2022?

daryelv, 2025-01-22:
Given that the developer has no interest in you leaving his ecosystem, he's not going to provide a way to do it.  Unfortunately this means you either 

1) buy all new devices - this would run me around $400-$500 for the devices plus all the time to replace the firmware on each new device with Tasmota, and finally all the time to physically replace the HAA device with the new Tasmota flashed one.  I'd estimate around 20-32 hours of labor to do this with the amount of devices and the locations they're in.

2) refresh existing devices - since you need to use hardware to reflash, this involves uninstalling the device, re-flashing, re-installing.  This isn't even possible on some devices that don't come apart (small outlets for example).  I'd estimate the hours to be around 48-56 total for all the devices in my setup.

3) a developer here embraces "Open Software" standards or provides a way to flash Tasmota over HAA.  Even if it was an old version of Tasmota that was 'blessed' with the signed keys that are required for flashing, that would at least let us reflash something like a ceiling fan control (I have 2) which are very difficult to get to.  

Considering this disclaimer was not added until July 14, 2022, I feel that a tool to allow flashing HAA back to Tasmota should be made available to anyone that has been stuck using HAA since before July 14, 2022. 

[image]