RavenSystem/esp-homekit-devices #1889: the light flickers above version 11.0.0

belkone, 2022-09-29:
Hi,
I have a problem with dimming/brightening and switching on/off - The light is flickering.
I am using an LED driver with the following MEPLHAA Script:
`{"c":{"q":600,"b":[{"g":0,"t":5}]},"a":[{"t":30,"g":[5,13],"b":[{"g":0}]}]}`

Up to version 11.0.0 it worked fine. Then the problems started. I tested the following HAA versions:
11.3.1 - works wrong(flickering)
11.2.0 - works wrong(flickering)
11.1.2 - works wrong(flickering)
11.1.1 - not available for installation
11.1.0 - not available for installation
11.0.0 - works fine

belkone, 2022-09-29:
thanks for the information :) I'm waiting for the new version and then I let you know if the problem is resolved :)

RavenSystem, 2022-09-30:
Fixed in https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_11.3.2

closed 2022-09-30 by RavenSystem

belkone, 2022-09-30:
I confirm, everything works fine again, thanks! :)