RavenSystem/esp-homekit-devices #1908: Toggle switch on Shelly 1 does not trigger anything

tokamac, 2022-11-02:
**Detailed steps to reproduce it**

I use a Shelly 1 v3 inside a wall switch box with two different circuits from the mains: 

- 1st circuit comes from a power socket available underneath, that powers the device. Wired to L and N connectors on the Shelly. 
- 2nd circuit is the original wall switch circuit with no neutral. Connected remotely to a ceiling lamp with live and hot return wires available in the switch box. The live wire is connected to 'I' and the hot return wire to the lamp is connected to 'O' on the Shelly.

This is possible because the Shelly 1 has I/O output dry contacts separated and isolated from its input power supply on L & N, so two different currents can flow in this device. 

This setup works perfectly in HomeKit. **But the physical toggle switch on the wall does not work.** It doesn't trigger anything.

I use the MEPLHAA below, the one intended for toggle switches. Tried to wire the toggle switch between L and SW, then between I and SW to no avail, nothing happens. Like this:

[image]

and

[image]

The switch highlighted in yellow does not work for one or the other.
Obviously I cannot group all the hot wires from the power socket and from the lighting circuit, otherwise the circuit breaker will trip.

**MEPLHAA Script**
`{"c":{"b":[{"g":5,"t":5}]},"a":[{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},   {"g":5,"t":0}]}]}
`

**Used device**
Shelly 1 v3

**HAA version**
11.4.2 Peregrine

**iOS/iPadOS version**
iOS 16.1

RavenSystem, 2022-11-03:
Check the hardware. I use second schema without any issue (Shelly 1 v3).

tokamac, 2022-11-07:
Thanks for the clarification, figure 2 then. But I triple-checked physical connections and the switch itself with a multimeter. Everything is fine, but alas the toggle doesn't switch the light on or off.

I followed figure 2 above and replaced the toggle switch with a push button (momentary switch) using the appropriate script:
`{"a":[{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0,"i":1,"t":0}]}]}`

But it's the exact same thing: the Shelly module works in the Home app and with Siri, but the physical button does nothing.

closed 2023-01-21 by stale[bot]