RavenSystem/esp-homekit-devices #1912: How to update and go to setup mode through home assistant?

alialki, 2022-11-13:
I am both new to HAA and Home assistant could anyone tell me how I can update firmware through home assistant homekit controller, as the devices are no longer exposed to HAA manager app through the bridge i created there is an entity for update defined as switch but when pressed nothing happens at all. 

I would really appreciate it if some one can help me.

ondratan, 2022-11-13:
Have you tried going to device setup and adding the device again?

alialki, 2022-11-13:
To be honest i wanted to avoid that route for the time being as it will be really time consuming

euhkah, 2022-11-23:
Hello, I'm looking for the same solution.

Automations in homekit are for the moment without interest, without conditions.

My devices use HAA firmware, but they are configured with Home Assistant.
I bought the HAA Manager application, but I cannot activate Setup Mode and update his.

It is necessary to update HAA Manager to make this possible, it is very important ! @RavenSystem

euhkah, 2022-11-23:
[In addition to the problem of unable to update, there is also the inability to modify the wifi configuration, the MEPLHAA configuration, etc.]

alialki, 2022-11-23:
I bought the HAA Manager app as well, I hope @RavenSystem updates the HAA manager to make this possible too. At the moment the best solution I found is to toggle the the device 8 times as fast as possible to take the devices in to setup mode then the app setup radar feature pops it up and you can open it in safari and change settings and update firmware from there. But still this is not an efficient solution specially for light switches, it will lower the light bulbs life span

alialki, 2022-12-02:
@euhkah have you managed to find a better solution than mine ? Because with my workaround i have blown two light bulbs since 10 days ago

euhkah, 2022-12-02:
I couldn't find anything better. I applied your solution to switch my lights to Homekit... temporarily. I'm thinking of going back to Tasmota, even if it means plugging in the wires. Especially since there is no form of support or help here.

freedreamer82, 2022-12-19:
it should be a very nice feature.
I also bought the app and every time I have to remove from HA the device , pair it back to my phone and update.... :( very long process ....
Please make it upgradable from  HA ( even if I I suspect is a problem HA side).

Marco

euhkah, 2022-12-19:
Hi Marco, as alialki said, you turn the device on/off several times. It will go into setup mode. By clicking on the app's radar, you can see your device, open its configuration and update it, without deleting it... But this procedure is also very long !

freedreamer82, 2022-12-20:
@euhkah no way , it's better haa than tasmota.
actually on home assistant there is a buttton to go on update setup mode the device...but i cannot get it working...
https://github.com/home-assistant/core/pull/59750

freedreamer82, 2022-12-22:
@euhkah I fixed the HA stuff.. once I'll be ready I'll make a PR on hass to have the chance to have button for UPDATE/SETUP in hass UI.
Regards 

Marco

freedreamer82, 2022-12-23:
@euhkah https://github.com/home-assistant/core/pull/84502
here you can find the mod if you wanna try.
cheers