RavenSystem/esp-homekit-devices #1914: Tuya TYJW2S modul integration

djsomi, 2022-11-15:
Hi All,

I recently got a wall mounted smart heater, which is equipped with this TYJW2S Tuya module. 
Does anyone tried to flash this or similar ones? My concern is that this module is controlling the unit on pin 1-2  RX & TX, so it is not so obvious (right now) for me how to get started.

Fortunately tuya has a page for this module, but

Any hint is appreciated!

Thanks
Zoltan

tonysprenk, 2022-11-15:
You can flash by wire or use tuya-convert wirelessly (if supported).  Make sure to make a backup of the current firmware in case something fails. 

When you have flashed with HAA you can use free monitor to control the secondary MCU.

djsomi, 2022-11-15:
Thanks, that's a goof starting point, I just have to figure out, the communication between the secondary MCU.

tonysprenk, 2022-11-15:
The tuya documentation is a good start for figuring out the commands. Also once you have flashed you can use free monitor to monitor the commands sent by the secondary mcu to the esp when you press the physical buttons for example.

closed 2023-01-21 by stale[bot]