RavenSystem/esp-homekit-devices #1936: Cannot access setup moe

rithask, 2022-12-28:
**Detailed steps to reproduce it**
I cannot access setup mode no matter which method I try. The HAA Wifi pops up after pressing the flash button for 8 seconds but when connecting, it just won't connect and disappears after showing an error in the iOS WiFi settings. I tried the double press method, pressing for 5 seconds. Both didin't help.

**MEPLHAA Script**
```json
{
    "c": {
        "l": 13,
        "ct": 10,
        "b": [
            {
                "g": 0,
                "t": 5
            }
        ],
        "m": 15
    },
    "a": [
        {
            "t": 24,
            "n": 2,
            "g": 14,
            "j": 10
        }
    ]
}
```

**Used device**
ESP8266

**HAA version**
v11.7.1

**iOS/iPadOS version**
16.2

**Logs**
A .txt file with logs. Read [how to capture logs from your device](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs)

**Screen-shots or videos showing issue**

SyBeer, 2023-01-01:
From my experience, probably device is broken. 
I have the similar observation with some sonoff basic and shelly devices.
 
What's interesting - theres no problem with connection to the device by GPIO pins to flash it, but devices can't operate normally.

rithask, 2023-01-02:
> From my experience, probably device is broken. 

The ESP8266 is new tho

Kienz, 2023-01-10:
Remove the "m" option - it's not needed.
Is your device connected to your Wifi? If yes, it should not open HAA Wifi hotspot. After enabling Setup Mode you have to enter <IP-Address>:4567 into the browser. Than the Setup Mode page should be displayed.
If it still doesn't work - please add logs.

rithask, 2023-01-10:
I removed that option and still can't access the setup mode by visiting IP:4567. The device is connected to WiFi, yes.

```
0027.365 192.168.1.213-FF0771: SNI..OK
0027.366 192.168.1.213-FF0771: SSLConnect..OK
0027.372 192.168.1.213-FF0771: Write..OK
0029.806 192.168.1.213-FF0771: GOT 2878 Bytes
0029.807 192.168.1.213-FF0771: HTTP/1.1 302 Found
0029.807 192.168.1.213-FF0771: Server: GitHub.com
0029.807 192.168.1.213-FF0771: Date: Tue, 10 Jan 2023 12:23:51 GMT
0029.807 192.168.1.213-FF0771: Content-Type: text/html; charset=utf-8
0029.807 192.168.1.213-FF0771: Vary: X-PJAX, X-PJAX-Container, Turbo-Visit, Turbo-Frame, Accept-Encoding, Accept, X-Requested-With
0029.816 192.168.1.213-FF0771: Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/166993980/deb9a7b9-3536-487f-8635-5118e017ba04?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230110%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230110T122351Z&X-Amz-Expires=300&X-Amz-Signature=298136ca9adf0646ca35637abc0ea27e79270a837dfe803bb9ed80c368a06c93&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
0029.864 192.168.1.213-FF0771: Cache-Control: no-cache
0029.866 192.168.1.213-FF0771: Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
0029.873 192.168.1.213-FF0771: X-Frame-Options: deny
0029.875 192.168.1.213-FF0771: X-Content-Type-Options: nosniff
0029.877 192.168.1.213-FF0771: X-XSS-Protection: 0
0029.879 192.168.1.213-FF0771: Referrer-Policy: no-referrer-when-downgrade
0029.883 192.168.1.213-FF0771: Content-Security-Policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com objects-origin.githubusercontent.com www.githubstatus.com collector.github.com raw.githubusercontent.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events *.actions.githubusercontent.com wss://*.actions.githubusercontent.com online.visualstudio.com/api/v1/locations github-production-repository-image-32fea6.s3.amazonaws.com github-production-release-asset-2e65be.s3.amazonaws.com insights.github.com wss://alive.github.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com objects-origin.githubusercontent.com; frame-ancestors 'none'; frame-src viewscreen.githubusercontent.com notebooks.githubusercontent.com; img-src 'self' data: github.githubassets.com media.githubusercontent.com camo.githubusercontent.com identicons.github.com avatars.githubusercontent.com github-cloud.s3.amazonaws.com objects.githubusercontent.com objects-origin.githubusercontent.com secured-user-images.githubusercontent.com/ opengraph.githubassets.com github-production-user-asset-6210df.s3.amazonaws.com customer-stories-feed.github.com spotlights-feed.github.com *.githubusercontent.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/ secured-user-images.githubusercontent.com/; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/
0030.040 192.168.1.213-FF0771: Content-Length: 0
0030.041 192.168.1.213-FF0771: X-GitHub-Request-Id: 27C9:3F8D:3DEAC0:4E4941:63BD58D0
0030.046 192.168.1.213-FF0771:
0030.046 192.168.1.213-FF0771:
0030.047 192.168.1.213-FF0771:
0030.047 192.168.1.213-FF0771: HTTP returns 302
0030.048 192.168.1.213-FF0771:
0030.067 192.168.1.213-FF0771: *** FORWARDING objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/deb9a7b9-3536-487f-8635-5118e017ba04?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230110%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230110T122351Z&X-Amz-Expires=300&X-Amz-Signature=298136ca9adf0646ca35637abc0ea27e79270a837dfe803bb9ed80c368a06c93&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain.bin.sec&response-content-type=application%2Foctet-stream
0030.105 192.168.1.213-FF0771: *** CONNECT
0030.164 192.168.1.213-FF0771: DNS..OK
0030.165 192.168.1.213-FF0771: Socket..OK
0030.225 192.168.1.213-FF0771: Connect..OK
0030.226 192.168.1.213-FF0771: SSL..OK
0030.226 192.168.1.213-FF0771: FD..OK
0034.280 192.168.1.213-FF0771: SNI..OK
0034.281 192.168.1.213-FF0771: SSLConnect..OK
0034.302 192.168.1.213-FF0771: Write..OK
0036.906 192.168.1.213-FF0771: GOT 828 Bytes
0036.906 192.168.1.213-FF0771: HTTP/1.1 206 Partial Content
0036.907 192.168.1.213-FF0771: Connection: keep-alive
0036.907 192.168.1.213-FF0771: Content-Length: 2
0036.907 192.168.1.213-FF0771: Content-Type: application/octet-stream
0036.907 192.168.1.213-FF0771: Content-MD5: n6zZPyzlgIY15vat6lDvmQ==
0036.910 192.168.1.213-FF0771: Last-Modified: Wed, 04 Jan 2023 22:11:46 GMT
0036.914 192.168.1.213-FF0771: ETag: "0x8DAEEA0AB787533"
0036.916 192.168.1.213-FF0771: Server: Windows-Azure-Blob/1.0 Microsoft-HTTPAPI/2.0
0036.921 192.168.1.213-FF0771: x-ms-request-id: 277695a6-901e-0044-1fee-24a85a000000
0036.926 192.168.1.213-FF0771: x-ms-version: 2020-04-08
[... cut ...]
0229.210 192.168.1.213-FF0771: 0x8B - 565248
0229.835 192.168.1.213-FF0771: 0x8C - 569344
0230.466 192.168.1.213-FF0771: 0x8D - 573440
0231.182 192.168.1.213-FF0771: 0x8E - 577536
0231.807 192.168.1.213-FF0771: 0x8F - 581632
0232.376 192.168.1.213-FF0771: 0x90 - 582068
0232.379 192.168.1.213-FF0771: >>> Verifying
0246.179 192.168.1.213-FF0771: >>> Result OK
0246.179 192.168.1.213-FF0771: *** FINISHING
0246.186 192.168.1.213-FF0771: * HAAMAIN 11.8.0 installed
0246.186 192.168.1.213-FF0771: *** REBOOTING
```

Kienz, 2023-01-10:
This is the logfile from update process. You updated the device to HAA 11.8.0.
Please start setup mode and post log.

rithask, 2023-01-11:
I figured it out. Misunderstood the wiki on accessing the setup mode. I should first press and hold the button for 8 seconds and then visit IP:4567.
Thanks for helping me out.

closed 2023-01-11 by rithask