RavenSystem/esp-homekit-devices #1940: Upgrade problem from HAA 5.x to current version

SyBeer, 2023-01-01:
**Detailed steps to reproduce it**
Hi, I found old Sonoff 4ch PRO R3 with HAA 5.0.18. I try by hard to upgrade it by OTA, but in doesn't work. 
I even setup my own HAA Upgrade Server with HAA 6.X and other version, but it doesn't work.

I can reach device from HAA Manager APP and switch it to Setup Mode. 

**MEPLHAA Script**
{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}],"m":120},"a":[{"0":{"r":[{"g":12,"v":1,"i":600}]},"1":{"r":[{"g":12}]},"t":4,"i":600,"b":[{"g":0}],"s":1},{"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"i":0.5,"b":[{"g":9}],"s":0},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"i":0.5,"b":[{"g":10}],"s":0},{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"i":0.5,"b":[{"g":14}],"s":0}]}

This script (works perfectly) is for Somfy Outdoor Blinds, and it need to specified wiring and integration in to original remote control . 
- First chanel is a security lock (it prevents from unwanted/unexpected blinds open operation)
- Second - Open linds
- Third - Stop operation 
- Fourth - Close blinds

**Used device**
Sonoff 4ch PRO R3

**HAA version**
5.0.18

**iOS/iPadOS version**
16.2 iOS

Kienz, 2023-01-10:
Logs?

valera16f84, 2023-01-24:
Same problem with my old Sonoff r2 with HAA 3.x. Auto update starts and finish again and again. Install new firmware via cable solve this problem))

tokamac, 2023-02-04:
> if you have a fresh version installed, then the update process is painless

If only it were true…
I use HAA Manager (the iOS app). All of my devices currently use Home Accessory Architect v11.7 Peregrine, a very recent release. Mainly Shelly and Sonoff devices, but also ESP8266-enabled electric sockets. _None of them can update since then._ Because none of them can enter setup mode anymore ("Setup Mode: Unreachable"). Every. Single. One. 
Restarting them does not change anything. Although they all work fine in Home! What happened, I'm still wondering.

tokamac, 2023-02-04:
Thanks for pointing this out, [xrust83](https://github.com/xrust83). That explains it all.

RavenSystem, 2023-02-06:
HAA Home Manager v3.4.4 is now ready at App Store.

closed 2023-05-01 by RavenSystem