RavenSystem/esp-homekit-devices #1966: can't get to setup mode, update, reboot or wifi reconnection using the app

toshibochan, 2023-02-05:
firmware 11.9.1
I'm using the HAA Manager to trying get to setup mode but not working. I don't know if is the app or the new firmware.i try all setup mode, update, reboot or wifi reconnection but not working.

InJecTeD-, 2023-02-05:
Same here.
First thought firmware is to old (11.7.1).
Erased (blank file) one of my shellys and flashed 11.9.1 .
Still no setup mode etc. using HAA Manager. 
Log shows "Upd CH" - nothing more. 
I think one of the last app updates broke it.

RavenSystem, 2023-02-05:
It is a known bug. It is fixed in v3.4.4, but I am waiting Apple approval.

RavenSystem, 2023-02-06:
HAA Home Manager v3.4.4 is now ready at App Store.

spacemannz, 2023-02-07:
Hi, the issue seems to persist for me with 3.4.4 - same as described above.

InJecTeD-, 2023-02-08:
Strange, for me 3.4.4 fixed it. (thank you @RavenSystem)

closed 2023-05-01 by RavenSystem