RavenSystem/esp-homekit-devices #1989: Migration Merlin

TheReiner, 2023-04-25:
i don't understand your migration... and all my devices is KO
And your example Database is not update ^^ Not yet i think ;) 
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates/b3bf97450019c69dfc7856e319be71485228becf

Exemple 1 (Led Ruban)
{"c":{"q":500,"b":[{"g":0,"t":5}],"u":1,"n":"Stub_evier"},"a":[{"t":30,"s":5,"g":[5,12],"b":[{"g":0}]}]}

My Output is GPIO 0 or 5 12 ? 
{"c":{"io":[[0],2],"q":500,"b":[{"g":0,"t":5}],"u":1,"n":"Stub_evier"},"a":[{"t":30,"s":5,"g":[5,12],"b":[{"g":0}]}]}

Exemple 2 (Sonof Mini) Oupput GPIO 12 and Input GPIO 4 ?
{"c":{"io":[[[12],2],[4],1]],"l":13,"b":[{"g":0,"t":5}],"u":1,"n":"Plafonnier_Cellier"},"a":[{"t":1,"s":5,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]}]}

Can you HelpMe ?
Thanks

TheReiner, 2023-04-25:
Ok i find my error for Sonoff Basic Mini ^^

{"c":{"io":[[ [ 12 ], 2 ],[ [ 0,4 ], 6 ]],"l":13,"b":[{"g":0,"t":5}],"u":1,"n":"Plafonnier_Cellier"},"a":[{"t":1,"s":5,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]}]}

TheReiner, 2023-04-25:
Ok for my Led Ruban... i don't find.

{"c":{"io":[[5],2],"q":500,"b":[{"g":0,"t":5}],"u":1,"n":"Stub_evier"},"a":[{"t":30,"s":5,"g":[5,12],"b":[{"g":0}]}]}

This is your migration guide..  if i read your example, and with my configuration. GPIO is 5 and mode output 2 ? Correct ?

A Lightbulb Service {"t":30,"ty":8,"g":[5,1,6]} needs the following declaration into "io", declaring GPIO 5 as output:

"io":[
  [ [ 5 ], 2 ]
]

RavenSystem, 2023-04-25:
Database is updated at https://github.com/RavenSystem/esp-homekit-devices/wiki/Devices-Database

This link https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates/b3bf97450019c69dfc7856e319be71485228becf is for a fixed commit.

This Lightbulb Service {"t":30,"ty":8,"g":[5,1,6]} is different than yours. This is NRZ, and yours is PWM:

{"c":{"io":[[[5,12],7],[[0],6]],"q":500,"b":[{"g":0,"t":5}],"u":1,"n":"Stub_evier"},"a":[{"t":30,"s":5,"g":[5,12],"b":[{"g":0}]}]}

TheReiner, 2023-04-25:
Many Thanks, it works fine ^^

closed 2023-04-25 by TheReiner