RavenSystem/esp-homekit-devices #1990: Updated using the HAA manager and lost use of lights

mrmark1235, 2023-04-25:
I just updated to HAA 12. Was done automatically using the app. All say updated but now my lights won’t turn on. Still appear in HomeKit and show them turning off and on but the light doesn’t actually come on… literally have so many and no lights in rooms. It’s a nightmare.

I’m not tech savvy at all so have very little clue what is wrong. Is it something to do with the script?

here’s the current scripts.

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]}]}

RavenSystem, 2023-04-25:
https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_12.0.0

```json
{"c":{"io":[[[13],3],[[0,4],6],[[12],2]],"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]}]}
```

closed 2023-04-25 by RavenSystem

mrmark1235, 2023-04-25:
Thank you! That has fixed them. I have one light which I reset the settings on and connected it back to Wi-Fi but I’m not seeing it in the HAA manager app or on Wi-Fi. How can I bring it back?