RavenSystem/esp-homekit-devices #1994: Roller Shutter on shelly 2.5 (Merlin v12) temperature problem

freedreamer82, 2023-04-26:
Hi,

I have 8 roller shutter running HAA with HW shelly 2.5 .I was pretty happy about Peregrine version, in fact everything was OK and the temperature of the device was at 35 degree.
After the update to merlin something went wrong...I updated just 2 devices and I noticed that temperature with merlin is 10 degree higher! now is about 45 (versus 35)

Peregrine script was:
{"c":{"n":"DEV1","l":0,"b":[{"g":2,"t":5}]},"a":[{"t":45,"o":21,"f":40,"c":21,"0":{"r":[{"g":15,"v":1,"i":0.8},{"g":4}]},"1":{"r":[{"g":15},{"g":4,"v":1,"i":0.8}]},"3":{"r":[{"g":15,"v":1,"i":0.8}]},"4":{"r":[{"g":4,"v":1,"i":0.8}]},"5":{"r":[{"g":4,"v":1,"i":0.3},{"g":15}]},"6":{"r":[{"g":15,"v":1,"i":0.3},{"g":4}]},"f0":[{"g":5,"i":1,"p":0,"t":0}],"f1":[{"g":13,"i":1,"p":0,"t":0}],"f2":[{"g":5},{"g":13}],"es":[{"t":22,"h":1,"n":5,"j":30,"y0":[{"v":75,"r":1,"0":{"m":[[1,-1]]}}]}]}]}
 
and with merlin (following the migration):
{"c":{"io":[[[0],3],[[2],6],[[5,13],6,0,1],[[4,15],2]],"n":"DEV1","l":0,"b":[{"g":2,"t":5}]},"a":[{"t":45,"o":21,"f":40,"c":21,"0":{"r":[{"g":15,"v":1,"i":0.8},{"g":4}]},"1":{"r":[{"g":15},{"g":4,"v":1,"i":0.8}]},"3":{"r":[{"g":15,"v":1,"i":0.8}]},"4":{"r":[{"g":4,"v":1,"i":0.8}]},"5":{"r":[{"g":4,"v":1,"i":0.3},{"g":15}]},"6":{"r":[{"g":15,"v":1,"i":0.3},{"g":4}]},"f0":[{"g":5,"t":0}],"f1":[{"g":13,"t":0}],"f2":[{"g":5},{"g":13}],"es":[{"t":22,"h":1,"n":5,"j":30,"y0":[{"v":75,"r":1,"0":{"m":[[1,-1]]}}]}]}]}

@RavenSystem  do you think could be a fw  bug ?  something wrong in the migration (maybe some GPIO status) ?  otherwise i have no clue....

For now I stopped updating all other devices 

Thanks for the support

           Marco

RavenSystem, 2023-04-26:
Could you try this?

{"c":{"io":[[[0],3],[[2],6],[[5,13],6,0,1],[[4,15],2],[[16],1]],"n":"DEV1","l":0,"b":[{"g":2,"t":5}]},"a":[{"t":45,"o":21,"f":40,"c":21,"0":{"r":[{"g":15,"v":1,"i":0.8},{"g":4}]},"1":{"r":[{"g":15},{"g":4,"v":1,"i":0.8}]},"3":{"r":[{"g":15,"v":1,"i":0.8}]},"4":{"r":[{"g":4,"v":1,"i":0.8}]},"5":{"r":[{"g":4,"v":1,"i":0.3},{"g":15}]},"6":{"r":[{"g":15,"v":1,"i":0.3},{"g":4}]},"f0":[{"g":5,"t":0}],"f1":[{"g":13,"t":0}],"f2":[{"g":5},{"g":13}],"es":[{"t":22,"h":1,"n":5,"j":30,"y0":[{"v":75,"r":1,"0":{"m":[[1,-1]]}}]}]}]}

freedreamer82, 2023-04-26:
> Could you try this? [...]

yes, much better, It was the irq pin not initialized i guess. Please update also the examples in the devices DB.

thanks 

Marco

closed 2023-04-26 by RavenSystem

RavenSystem, 2023-04-26:
I updated a Shelly 2.5 to v12.0.1, and temperature is right again. Many thanks!