RavenSystem/esp-homekit-devices #1998: Hello. After updating iFan03 to version 12, it stopped working normally. In the Home Kit, the speed is adjustable – in reality, the fan immediately turns on 100% and cannot be adjusted. (always works 100%). What did I do wrong? It looks like the configuration in the Devices Database is incorrect.

fiska1, 2023-04-30:
**Detailed steps to reproduce it**

**MEPLHAA Script**

**Used device** 

**HAA version**

**iOS/iPadOS version**

**Logs**
A .txt file with logs. Read [how to capture logs from your device](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs)

**Screen-shots or videos showing issue**

fiska1, 2023-04-30:
Wrong job!!! Configured from the remote, the speeds work correctly. When using the remote control, the 1st speed is 3,333%; the 2nd speed is 2,000%; the 3rd speed is 1,000%. When working in Home Kit 33% 67% and 100% !!! The hood immediately turns on to the maximum and is not regulated! Please fix the work in JSON

fiska1, 2023-04-30:
Wrong job!!! Configured from the remote, the speeds work correctly. When using the remote control, the 1st speed is 3333%; the 2nd speed is 2000%; the 3rd speed is 1000%. When working in Home Kit 33% 67% and 100% !!! The hood immediately turns on to the maximum and is not regulated! Please fix the work in JSON

RavenSystem, 2023-05-01:
I don't understand you, and more information is needed. No logs and no script provided.