RavenSystem/esp-homekit-devices #2008: iFan03 Incorrect work

fiska1, 2023-05-01:
1) When using the Home Kit, % shows correctly, but the fan immediately turns on at maximum speed and is not adjustable. Photos 1, 2, 3,
2) When using the remote control, the speed is regulated (vice versa), 1- maximum, 3-minimum. At the same time, the slider in the Home Kit immediately shows the maximum, % of the speed shows 1000 , 2000 , 3333 . Photo 4,5,6.
3) JSON used with Devises Database.
{"c":{"io":[[[13],3],[[0],6],[[9,12,14,15],2]],"z":12,"r":[{"n":10,"s":9600,"l":[8,8]}],"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":65,"0":{"r":[{"g":15},{"g":12},{"g":14}]},"y0":[{"v":1,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":12},{"g":14,"i":0.6}]}},{"v":33,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":12,"i":0.6},{"g":14,"i":0.6}]}},{"v":67,"0":{"r":[{"g":15,"v":1},{"g":12},{"g":14,"v":1}]}}],"es":[{"s":1,"1":{"r":[{"g":9}]},"0":{"r":[{"g":9,"v":1}]},"b":[{"g":0}]}]},{"t":80,"h":0,"n":24,"pt":[["AA",5]],"dt":[2,0],"y0":[{"v":260,"r":1,"0":{"m":[[7002,1]]}},{"v":517,"r":1,"0":{"m":[[7002,1]]}},{"v":6,"r":1,"0":{"m":[[7001,1],[-2]]}},{"v":263,"r":1,"0":{"m":[[7001,1],[-2,30],[-2,101]]}},{"v":520,"r":1,"0":{"m":[[7001,1],[-2,60],[-2,101]]}},{"v":777,"r":1,"0":{"m":[[7001,1],[-2,100],[-2,101]]}},{"v":1034,"r":1,"0":{"m":[[7001,1],[-1,4]]}},{"v":265,"r":1,"0":{"m":[[7004,1]]}},{"v":266,"r":1,"0":{"m":[[7003,1]]}}]},{"h":0,"i":0.1,"1":{"u":[{"v":"AA550104000005"}]}},{"h":0,"i":0.1,"1":{"u":[{"v":"AA550101000002"}]}},{"h":0,"i":0.1,"1":{"u":[{"v":"AA550107000008"}]}},{"h":0,"i":0.1,"1":{"u":[{"v":"AA550106000007"}]}}]}
![3](https://user-
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
images.githubusercontent.com/30092533/235448071-5f1ccbb1-1902-48c3-8768-24c1ea37c635.jpg)

fiska1, 2023-05-01:
On HAA v.11, everything worked correctly with such JSON
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":65,"e":3,"0":{"r":[{"g":15},{"g":12},{"g":14}]},"y0":[{"v":3,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":12},{"g":14,"i":0.6}]}},{"v":2,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":12,"i":0.6},{"g":14,"i":0.6}]}},{"v":1,"0":{"r":[{"g":15,"v":1},{"g":4},{"g":14}]}}]},{"1":{"r":[{"g":9}]},"0":{"r":[{"g":9,"v":1}]},"b":[{"g":0}]}]}

RavenSystem, 2023-05-01:
Your script has different HomeKit services. You must use same services with same % configuration to avoid issues if you don't want to re-pair it to HomeKit. Use this v12 Merlin updated script base on your old script:

```json
{"c":{"io":[[[13],3],[[0],6],[[15,12,14,9],2]],"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":65,"e":3,"0":{"r":[{"g":15},{"g":12},{"g":14}]},"y0":[{"v":3,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":12},{"g":14,"i":0.6}]}},{"v":2,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":12,"i":0.6},{"g":14,"i":0.6}]}},{"v":1,"0":{"r":[{"g":15,"v":1},{"g":4},{"g":14}]}}]},{"1":{"r":[{"g":9}]},"0":{"r":[{"g":9,"v":1}]},"b":[{"g":0}]}]}
```

RavenSystem, 2023-05-01:
Please, use issues only for firmware issues.

closed 2023-05-01 by RavenSystem