RavenSystem/esp-homekit-devices #2046: Pairing connection issue

borissov, 2023-06-03:
**Detailed steps to reproduce it**

When I try to pair, the connection breaks. I deleted it, updated it, reset the HomeKit ID several times, manually set the IP in the router, but nothing helps. I even turned off all the other devices.

Video of reproducing:
https://www.youtube.com/watch?v=z6NJfKEWTZk

**MEPLHAA Script**
{
  "c": {
    "io": [
      [
        [
          12,
          13
        ],
        2
      ],
      [
        [
          0
        ],
        6
      ]
    ],
    "l": 13,
    "m": 60,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1
          }
        ]
      },
      "t": 2,
      "b": [
        {
          "g": 0
        }
      ]
    }
  ]
}

**Used device**

Sonoff S26

**HAA version**

12.4.4

**iOS/iPadOS version**

12.5 and 16.4.4

VaBanck, 2023-06-04:
`{"c":{"io":[[[13],3],[[0],6],[[12],2]],"l":13,"m":60,"b":[[0,5]]},"a":[{"0":{"r":[[12]]},"1":{"r":[[12,1]]},"t":2,"b":[[0]]}]}`

borissov, 2023-06-04:
And with this configuration, it reacts in the same way again.

RavenSystem, 2023-06-04:
I tested it, and works fine. Check if your device is broken.

mariusvasss, 2023-06-04:
I have the same issue, 11.8 works perfectly and then update…

borissov, 2023-06-04:
> 

Device respond the pings, setup mode works, button works and switch is clicking.. I have 6 of this Sonoffs.. I think some watchdog probably kill the esp on pairing process or god knows what happens :(.

RavenSystem, 2023-06-04:
Logs are needed.

mariusvasss, 2023-06-04:
All new devices flashed with full haa boot

borissov, 2023-06-04:
All of mine are upgraded from 11 with reset Homekit IDs because I had to register a new home.

mariusvasss, 2023-06-04:
It gives you the error ‘unable to add device’?

borissov, 2023-06-04:
Yes, can’t pair it. I think device reset before end of pairing, ping is missing and led blink like reconnect to wifi.

mariusvasss, 2023-06-04:
Same to me. I have 11.8 saved on my pc and I flash them with it and then i update…. Esp32 works fine to pair, only esp8285/8266 gives this error… 
Flash with older firmware and update after….

mariusvasss, 2023-06-05:
Today i have changed an accesory and forgot to remove from home, i made it switches and before was blinds… after reset homekit id, i can’t pair it again in home…. ‘Unable to add accessory!’

RavenSystem, 2023-06-05:
Again: I can not reproduce the issue. Logs are needed. And to know specific version where this issue appears is useful too, because  I can not reproduce the issue.

mariusvasss, 2023-06-05:
Uart logs capture?

RavenSystem, 2023-06-05:
Yes, and knowing exact version where issue appears will be very helpful.

Starksoft, 2023-06-08:
@borissov Try to reset your router. Sometimes mDNS resolution fails in local network. On my Asus RT-AC66U this is common issue.

Also try to monitor your network with network monitoring app like Debookee (https://debookee.com/)
You'll see all your devices and mDNS names assigned to them.

borissov, 2023-06-08:
I check it with "Discovery - DNS - SD Browser" app for IOS, mDNS records is OK.

borissov, 2023-06-08:
After resoldering the pins and flashing, I attempted to redownload the HAA firmware multiple times. I also tried rewriting the JSON configuration, and finally achieved success with the following JSON configuration:

```json
{
  "c": {
    "io": [
      [
        [
          12,
          13
        ],
        2
      ],
      [
        [
          0
        ],
        6
      ]
    ],
    "m": 60,
    "l": 13,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1
          }
        ]
      },
      "b": [
        {
          "g": 0
        }
      ]
    }
  ]
}
```

borissov, 2023-06-09:
Now I'm on the dev beta of iOS 17, and pairing works normally with two codes. It's probably an issue with iOS 16 on iPhone 11.

RavenSystem, 2023-06-12:
No logs and info about which version introduced issue?

closed 2023-06-13 by RavenSystem