RavenSystem/esp-homekit-devices #2089: Free Monitor I2C bugs

Tellicious, 2023-07-29:
I'm trying to set up a CO2 / temperature / humidity monitor using an ESP32-S2 (Wemos S2 mini) connected to a Sensirion SCD41 via I2C. While doing so I think I found some bugs in I2C implementation in HAA 12.7.1. The only configuration that seems to work is Free Monitor with `"n":20` and just 1 accessory.

Let me know if you need any further info to sort that out

- **Bug 1**: When using multiple Free Monitor with `"n":20` targeting the same device, it works fine BUT sends all the read commands one after the other, without any delay, causing the sensor to return the first value correctly and then NACK on the second and third read
Script:
```
{
    "c": {
        "io": [[[15],2],[[0],6,1,1]],
        "l": 15,
        "i": 0,
        "ic": [[9, 8, 90, 1, 1]],
        "v": 2,
        "o": 4,
        "b": [[0,5]],
        "n": "HAA-AirQuality"
    },
    "a": [
        {
            "t": 15,
            "dt": [4]
        },
        {"t": 10, "i": 60}, 
        {"t": 24},
        {
            "t": 80,
            "n": 20,
            "in": [[0, 33, 177]],
            "ic": [0, 98],
            "dt": [2, 0],
            "l": [400, 5000],
            "y0": [{"v": 1000, "r": 1, "0": {"m": [[2, 1], [2, -1]]}}]
        },
        {
            "t": 80,
            "n": 20,
            "tg": [3,0],
            "ic": [0, 98],
            "io": 3,
            "dt": [2, 0],
            "ff": 0.00267033,
            "fo": -45,
            "l": [-10, 60]
        },
        {
            "t": 80,
            "n": 20,
            "tg": [3,1],
            "ic": [0, 98],
            "io": 6,
            "dt": [2, 0],
            "ff": 0.0015259
        }
    ]
}
```
[image]
[image]

-**Bug 2**: When using even a single Free Monitor with `"n":21`, and with both `In` and `It`, it continuously sends 0x00 bytes for a second during startup, then a read command right after (according to the `It` delay) and then SDA is pulled low and nothing more happens on the bus
Script:
```
{
    "c": {
        "io": [[[15],2],[[0],6,1,1]],
        "l": 15,
        "i": 0,
        "ic": [[9, 8, 90, 1, 1]],
        "v": 2,
        "o": 4,
        "b": [[0,5]],
        "n": "HAA-AirQuality"
    },
    "a": [
        {
            "t": 15,
            "dt": [4]
        },
        {"t": 10, "i": 60}, 
        {"t": 24},
        {
            "t": 80,
            "n": 21,
            "in": [[0, 33, 177]],
            "it": [1, 0, 236, 5],
            "ic": [0, 98],
            "dt": [2, 0],
            "l": [400, 5000],
            "y0": [{"v": 1000, "r": 1, "0": {"m": [[2, 1], [2, -1]]}}]
        }
    ]
}
```
[image]
[image]

- **Bug 3**: When using even a single Free Monitor with `"n":21`, and with only `It`, it continuously sends 0x00 bytes for a second during startup as in Bug 2, then starts regularly sending a Write command and a Read command 5 seconds after (according to the delay specified in `It`) but totally forgets the other values specified in `It` that should be sent right after the Write command, regardless of what combination is used (`"It":[5,2,33,157]`, `"It":[5,0,33,157]`, `"It":[5,2,33,157,1,1]` they all produce the same result you can see below)
Script:
```
{
    "c": {
        "io": [[[15],2],[[0],6,1,1]],
        "l": 15,
        "i": 0,
        "ic": [[9, 8, 90, 1, 1]],
        "v": 2,
        "o": 4,
        "b": [[0,5]],
        "n": "HAA-AirQuality"
    },
    "a": [
        {
            "t": 15,
            "dt": [4]
        },
        {"t": 10, "i": 60}, 
        {"t": 24},
        {
            "t": 80,
            "n": 21,
            "in": [5, 0, 33, 157],
            "ic": [0, 98],
            "dt": [2, 0],
            "l": [400, 5000],
            "y0": [{"v": 1000, "r": 1, "0": {"m": [[2, 1], [2, -1]]}}]
        }
    ]
}
```
[image]
[image]

RavenSystem, 2023-07-29:
Please, try https://github.com/RavenSystem/haabeta/releases/tag/12.7.2b1

Tellicious, 2023-07-29:
Hi, I just made a test and it is showing the same behavior as before. Actually the setup still shows version 12.7.1 and the CRC32 looks the same for this beta and for the 12.7.1 release. Could it be that there was a mistake in the upload of the beta?

By the way I also have to correct a small mistake in the third script: I wrote `In` instead of `It`, but the one that was running was correctly with `It`

- **Bug 1**: no changes. I had one acquisition in which the clock slowed down significantly on the third read, but maybe it was just a random case. Anyway my suggestion would be to add a customizable delay as a parameter of each Free Monitor, maybe in a future release
[image]
- **Bug 2**: no changes
[image]
-**Bug 3**: no changes

RavenSystem, 2023-07-29:
I checked them, and even file size is different. You should see 12.7.2b1 in setup. How are you installing beta? You must follow this: https://github.com/RavenSystem/haabeta

About bug 1, there is a n option to set a delay after service is created to avoid overlapping tasks: https://github.com/RavenSystem/esp-homekit-devices/wiki/accessory-configuration#delay-after-creation

But I want to test without using it.

Tellicious, 2023-07-29:
Alright, apologies I must have missed the "cd" option. 

I was re-flashing fullhaaboot_esp32s2 from the beta release link. Using the custom OTA server it worked

This is the update:

- **Bug 1**: `in` command is issued correctly at startup but then it is issued again multiple times (with a delta of around 20 seconds). I did manage to catch 3 `in` commands being sent as I was just collecting 60s of data. Then the 3 read commands are issues one after the other. There is a 2ms delay between the first and the second and the third one is right next to the second one
[image]
[image]
[image]

- **Bug 2**: now the 0s during startup last for 2 seconds instead of 1. Then again a Read command is sent and SDA is pulled low. Even pressing the RST key, SDA remains low
[image]

- **Bug 3**: now the 0s during startup last for 2 seconds instead of 1. It's just a write command followed by several 0x00. Write command is again issued without any other command (using the script above) and the read command is issued 7 seconds after (configuration requires 5)
[image]
[image]
[image]

 Unfortunately I'll be traveling for the next 20 days so I think I'll be able to make some more tests only in the second half of August. Let me also know if you want me to test something specific

RavenSystem, 2023-08-01:
> Bug 1: in command is issued correctly at startup but then it is issued again multiple times (with a delta of around 20 seconds).

This is wrong, because `in` is processed only at boot. Something is causing device reboots. And I think all issues are not related directly with I2C, but with DRAM assigned to Free Monitor task. I will do some tests and report.

RavenSystem, 2023-08-03:
I see that you are using log out `"o":4`. This is unsafe and it an cause issues with tasks. You must use buffered version to avoid issues with I2C.

Tellicious, 2023-08-05:
Alright, I'll test again all the 3 cases without any logging and report back. Unfortunately I won't be able to do so before the 20th of August.
The reason why I was using the unbuffered logging was that the buffered one was mixing up messages every now and then, as if the buffer was not big enough and it was being partially overwritten before being printed.

RavenSystem, 2023-08-06:
You are right. Buffered logging does not work as expected. I am doing some improvements with this for the next revision. However, unbuffered logging can not be used because it creates instability on the system.

As far I know, I2C was working fine without unbuffered logging, but thanks to your issue, I have improved it.

Soon, I will release next revision 12.7.3 with bug fixes, performance improvements, and a little new feature.

Tellicious, 2023-08-23:
Alright, tested again v12.7.3 with `"o":8`
Here the update:
- **Bug 1:** starting command `in` is correctly issued once and subsequent reading operations are spaced by a random delta, that varies between 0 and 2.5 ms
- **Bug 2:**  no changes from what I previously reported apart that the sequence of `0x00` lasts for 1s instead of 2s
- **Bug 3:** no changes from what I previously reported apart that the sequence of `0x00` lasts for 1s instead of 2s and that the read command is issued between 5 and 6s after the empty write command (configuration still requires 5)

I have also to report that on ESP32-S2 with 12.7.3 it is now vary hard to enter setup mode via physical command. Once the button is press (long press) I see `Setup call` and `Rebooting` in the log, each of them followed by an empty message but the webpage doesn't load no matter what I try. Most of the times manually unplugging/re-plugging the device after the device tried to enter setup mode gets it to work, other times it reboots automatically into running mode after 20-30s or so and I have to repeat the setup process over and over several times before I find a way to access the webpage. I even tried stopping my Wi-Fi to connect via the integrated AP but, even if the AP network shows up fine, no device can connect to it.
The behavior seems to change also based on the uploaded script and I'm having a hard time replicating it. I also re-flashed from scratch the fullhaaboot.bin image but no changes

Tried an ESP8266 and everything works as usual

Tellicious, 2023-08-24:
I did some other tests today on the setup topic: actually it seems to be strongly related to the script. Using a super simple switch script, it works flawlessly while, with any of the 3 scripts above, it becomes problematic.

RavenSystem, 2023-08-24:
Have you got the HAA logs about I2C?

Tellicious, 2023-08-31:
Yep, sure. For some reason I didn't receive any notification of your comment

- **Bug 1:**
[Aug 31 222028 HAA Log.log](https://github.com/RavenSystem/esp-homekit-devices/files/12490088/Aug.31.222028.HAA.Log.log)

- **Bug 2:**
[Aug 31 222245 HAA Log.log](https://github.com/RavenSystem/esp-homekit-devices/files/12490089/Aug.31.222245.HAA.Log.log)

- **Bug 3:**
[Aug 31 222602 HAA Log.log](https://github.com/RavenSystem/esp-homekit-devices/files/12490091/Aug.31.222602.HAA.Log.log)

Let me know if I can help in any way or do some tests for you

RavenSystem, 2023-08-31:
Is there any I2C device connected to ESP?

Tellicious, 2023-08-31:
Yes, it's a Sensirion SCD41.

From my point of view the behavior  of Bug 1 script in 12.7.3 is fine: read commands are no longer sent one after the other but they do have a delay now.
The fact that the sensor isn't returning any value is according to expectation, as they are still too close to each other for it to return a value (a reading is provided every 5s). Adding a delay to the 5th and 6th service make it work flawlessly. I kept testing without it as you asked me to do so.

The scripts of Bug 2 and Bug 3, instead, should work fine but they don't, as what is happening on the I2C bus is not what should be happening according to the config script

RavenSystem, 2023-09-01:
I see `! <n> I2C` in logs. This is an I2C error. I am going to add more verbose logs to know what is happening in a new beta version.

Tellicious, 2023-09-01:
I think it's simply because the sensor is not returning any value. In both bugs 2 and 3 the IT command is not sent out by the ESP, thus the sensor doesn't receive the trigger and doesn't return any value at the _read_ command

In Bug 2 the SDA line is also pulled low by the ESP, so no communication can happen on that line

RavenSystem, 2023-09-06:
Please, try last beta 12.8.0:

https://github.com/RavenSystem/haabeta/releases/tag/12.8.0b1

Tellicious, 2023-09-08:
Alright, tested with 12.8.0b2
- **Bug 2**
Initialization is now performed correctly (Write+0x21+0xB1), however, the first trigger command is Write+0x00+0x00+.... for 1 second (image 2) instead of Write+0xEC+0x05 and subsequent trigger commands are just Write instead of Write+0xEC+0x05(image 3).
Delay between trigger command and following read is around 2s (image 3) instead of 1s as defined by the script
However SDA is no longer pulled low
[image]
[image]
[image]

Tellicious, 2023-09-08:
**Bug 3:**
Still starts with Write + 0x00 + 0x00 +... for 1 sec (shouldn't be writing anything according to the configuration script), then the trigger command is sent empty (image 2), just a Write (instead of Write+0x21+0x9D), and the read command is issues almost 6 sec after the Write (instead of 5 as per config) `"it": [5, 0, 33, 157]`
[image]
[image]

RavenSystem, 2023-09-08:
I found a bug in I2C trigger length. Try beta 3, please.

https://github.com/RavenSystem/haabeta/releases/tag/12.8.0b3

Tellicious, 2023-09-09:
Alright,
I tested again **Bug 3** and it seems to be solved now. I'll make a few more tests also on the other one and I'll let you know

Tellicious, 2023-09-10:
Tested also **Bug 2** and it seems solved now. I did however find another bug related to the timing. 

Configuration is 3 tasks all accessing the same bus and sensor, all with a trigger and a required wait time between trigger and read (0.1s or 5s). What happens is that the second tasks sends its trigger during the wait time of the first task, as if the bus was released as soon as the write operation is complete.

I'd suggest three ways to solve this:
1. Keeping the mutex until the read operation is complete (with something like a token-ring in place to avoid that only one task keeps the mutex in case of delays in `"it"` longer than `"j"`)
2. Specifying in the wiki that timing of the I2C sensors should be handled manually to avoid such situations
3. Automatically delaying creation of the following tasks by a tad more than the delay specified in `"it"` (still they could overlap in case of `"it"` delays comparable with `"j"`)

Images and scripts below 

```
{
	"c": {
		"io": [
			[
				[15], 2
			],
			[
				[0], 6, 11, 1
			]
		],
		"l": 15,
		"i": 0,
		"ic": [
			[9, 8, 90, 1, 1]
		],
		"v": 2,
		"o": 8,
		"b": [
			[0, 5]
		],
		"n": "HAA-AirQuality3",
		"ntp": "ntp1.inrim.it",
		"tz": "CET-1CEST-2,M3.5.0/2,M10.5.0/3"
	},
	"a": [{
		"t": 24,
		"es": [{
			"t": 10,
			"i": 60
		}]
	}, {
		"t": 80,
		"h": 0,
		"n": 21,
		"tg": [1, 0],
		"it": [0.1, 0, 33, 150],
		"ic": [0, 98],
		"io": 3,
		"dt": [2, 0],
		"ff": 0.00267033,
		"fo": -45,
		"l": [-10, 60]
	}, {
		"t": 80,
		"h": 0,
		"n": 21,
		"tg": [1, 1],
		"it": [0.1, 0, 33, 150],
		"ic": [0, 98],
		"io": 6,
		"dt": [2, 0],
		"ff": 0.0015259
	}, {
		"t": 80,
		"h": 2,
		"n": 21,
		"it": [5.1, 0, 33, 157],
		"ic": [0, 98],
		"dt": [2, 0],
		"l": [400, 5000],
		"y0": [{
			"v": 1000,
			"r": 1,
			"0": {
				"m": [
					[2, 1],
					[2, -1]
				]
			}
		}],
		"es": [{
			"t": 95,
			"h": [3, 0, 10],
			"j": 900,
			"x": 1
		}, {
			"t": 95,
			"h": [4, 0, 10],
			"j": 900,
			"x": 1
		}, {
			"t": 95,
			"h": [5, 0, 10],
			"j": 900,
			"x": 1
		}]
	}]
}
```
[image]
[image]

RavenSystem, 2023-09-10:
I added a check to skip I2C Free Monitor task creation if there is other Free Monitor I2C Trigger task running with same bus and address.

Try it:
https://github.com/RavenSystem/haabeta/releases/tag/12.8.0b5

Tellicious, 2023-09-10:
Just tested 12.8.0b5: it seems to have the same behavior as before:
[image]
[image]

Anyway, actually, in the example above I do need all 3 Free Monitor tasks:
1. First one sends trigger 0x2196, waits 0.1s and reads temperature from byte 4 and 5 of sensor answer
2. Second one sends trigger 0x2196, waits 0.1s and reads humidity from byte 7 and 8 of sensor answer
3. Third one sends trigger 0x219D, waits 5.1s and reads CO2 from byte 1 and 2 of sensor answer

Either there's a way to extract multiple data from the same sensor answer or all the 3 tasks should be running, completing their full trigger + read cycle before handing over the mutex to the following task

RavenSystem, 2023-09-10:
You must add "cd":0.5 to avoid overlapping task creation. HAA works with a task create and destroy system to save resources. A mutex can not be used because it leaves a task in a waiting state, wasting DRAM.

RavenSystem, 2023-09-13:
There was a bug in the I2C safeguard. Try next beta:

https://github.com/RavenSystem/haabeta/releases/tag/12.8.0b6

Without a logic analyzer, you can see HAA logs to see if there is any task overlap.

RavenSystem, 2023-09-14:
I released final v12.8.0 with all fixes and modifications. Many thanks for your help. I added an advise to Wiki about tasks overlapping too.

closed 2023-09-14 by RavenSystem