RavenSystem/esp-homekit-devices #2104: Lights randomly turn off

miguelpruivo, 2023-09-04:
This seems to happen randomly and no particular given time and, so far, I believe it only happened with my Shelly 2.5 that uses the double light switch template. It happened with more than one device so it's not malfunction hardware but I also can't provide the exact correct steps to reproduce, since it happens randomly with no apparent reason. Lights are on and randomly turn off like if the device had restart on its own (and since default option is off, the lights would turn off). There is no inductive load involved in those scenarios, it's just led lights. Devices doesn't seem to disconnect (at least my router doesn't report it) and temperatures are below 50º when this happens.

**MEPLHAA Script**
```
{"c":{"io":[[[4,15,0],2],[[2],6],[[5,13],6,0,1]],"l":0,"b":[[2,5]],"ic":[[14,12,100]]},"a":[{"0":{"r":[[4]]},"1":{"r":[[4,1]]},"b":[[13],[13,0]],"es":[{"0":{"r":[[15]]},"1":{"r":[[15,1]]},"b":[[5],[5,0]]},{"t":75,"cd":1.5,"n":2,"dt":[0,56],"vf":0.0000382602,"vo":-0.068,"cf":0.00000949523,"co":-0.017,"pf":0.006097560976},{"t":75,"n":3,"dt":[0,56],"vf":0.0000382602,"vo":-0.068,"cf":0.00000949523,"co":-0.017,"pf":-0.006097560976},{"t":22,"h":2,"n":5,"j":10,"y0":[{"v":75,"r":1,"0":{"m":[[1],[2]]}}]}]}]}
```

**Used device**
Shelly 2.5

**HAA version**
Latest

**iOS/iPadOS version**
Latest stable

**Logs**
N/A

**Screen-shots or videos showing issue**
N/A

RavenSystem, 2023-09-04:
Logs are needed.

miguelpruivo, 2023-09-05:
> Logs are needed.

How do I get logs from this? It happens randomly. Can happen like 2 days in a row or one month without happening. Is there any place where I can retrieve the logs? Or enable on device logs and then download OTA?

macjl, 2023-09-05:
Hello. I have the same issue here. 
I guess it’s related to WiFi, as it’s happen when WiFi signal seems bad (microwave running, or instability on my AP). Very difficult to provides logs.

miguelpruivo, 2023-09-05:
> Hello. I have the same issue here.
> I guess it’s related to WiFi, as it’s happen when WiFi signal seems bad (microwave running, or instability on my AP). Very difficult to provides logs.

Ok, if that's it, why would that make the device reboot/cycle between on/off or even turning off? Based on the router connection status the connection is maintained when this happens, so I assume that it doesn't even disconnect.

RavenSystem, 2023-09-09:
@macjl Your issue is different. A wifi disconnection/reconnection in an environment with interferences can cause device reboots. I try to mitigate this, but I can not fix it because wifi driver is closed source. v12.8.0 has more mitigations for your issue.

RavenSystem, 2023-09-09:
@miguelpruivo Without logs, it is very difficult to know what command is turning off the service.

miguelpruivo, 2023-09-09:
> @miguelpruivo Without logs, it is very difficult to know what command is turning off the service.

My issue can be similar to @macjl because I’m not sure if there isn’t interference or not when the light turns off.

I could provide you with logs but how can I capture the logs if this only happens randomly?

RavenSystem, 2023-09-09:
You must use a computer, capturing logs all time.

miguelpruivo, 2023-09-09:
> You must use a computer, capturing logs all time.

That isn’t doable because it can take weeks or months to happen. It’s random. But assuming it is because of the interference, how can one prevent it?

RavenSystem, 2023-09-14:
If issue is caused by interferences and wifi reconnections, causing device to reboot, your router should show a reset connected time. You can try it by yourself, unplugging and plugging again any HAA device and seeing what router shows.

closed 2023-10-06 by RavenSystem