RavenSystem/esp-homekit-devices #2124: AP Mode not working after losing network connection

Tellicious, 2023-10-01:
Steps:
1. Device configured and running, connected to Wi-Fi network
2. _Wi-Fi AP_ enabled on the device without any password set 
3. Wi-Fi network turned off

Result:
- Device doesn't enter into setup mode automatically and AP is not turned on
- Forcing it into setup mode via physical control turns on AP but it is not possible to connect to the _HAA-xxxx_ network (it seems it doesn't accept connections)
- Rebooting the device several times doesn't fix it

Device: ESP32-S2
HAA version: 12.8.2

RavenSystem, 2023-10-02:
AP will not appear when running normal mode, for security reasons. AP only works in setup mode. Bug is:
> Forcing it into setup mode via physical control turns on AP but it is not possible to connect to the HAA-xxxx network (it seems it doesn't accept connections)

It will be fixed with next version.

RavenSystem, 2023-10-04:
Try https://github.com/RavenSystem/haabeta/releases/tag/12.9.0b1

Tellicious, 2023-10-05:
Unfortunately, even if in 12.9.0b1 it started working, it is still showing a strange behavior:
- The first time I turn on the AP by entering into setup mode, I always receive the message "The network HAA-xxxx could not be joined"
- Sometimes, after a couple of connection attempts, the device reboots and, as soon as the AP appears again, I'm able to connect (if it appears again by itself. Other times I have to manually trigger setup mode 2-3 times before being able to connect)
- Sometimes, also, the first time I click "save" on the configuration page, I receive the message that the server stopped responding and, after a while, the configuration page is loaded again. It then works fine on the second "save" attempt

Tried several times today but I was not able to identify a repeating pattern

RavenSystem, 2023-10-06:
Entering setup mode will turn on HAA WiFi AP, but it will be turned off when device connects to WiFi router (unless you setup an AP Password).

Tellicious, 2023-10-06:
The behavior I was describing is happening when Wi-Fi is not available. I setup a dedicated network that I can pause and activate on demand just to test this

RavenSystem, 2023-10-06:
Try new beta, https://github.com/RavenSystem/haabeta/releases/tag/12.9.0b3

RavenSystem, 2023-10-08:
Try final release 12.9.0. But sometimes AP fails because device is searching for configured WiFi, and it can interrupt AP task, causing you need a new try. It is a SDK matter, and WiFi components are closed source in ESP-IDF, then I can not do much more.

closed 2023-10-08 by RavenSystem

Tellicious, 2023-10-09:
Looks like it improved significantly in latest release, even though sometimes it still fails. Well understood the issue with the SDK; anyway, thanks a lot for taking time investigating the issue