RavenSystem/esp-homekit-devices #2125: Wrong log output for UART action

Tellicious, 2023-10-01:
Description:
Log output for an UART action looks like each byte is duplicated and without the last HEX value, even though the command is sent out correctly:
- Command: `16110b000000070000036b0000001202000003a`
- Log output: `161611110b0b00000000000007070000000003036b6b00000000000012120202000000000303`

Device: ESP32-S2
HAA version: 12.8.2
Script extract of accessory 20:
```
{
"t": 80,
"n": 5,
"h": 0,
"dt":[0,13,0],
"j": 5,
"y0": [{"v": 0, "r": 1, "0": {"u":[{"n":0, "v": "16110b000000070000036b0000001202000003a"}]}},
           {"v": 3, "r": 1, "0": {"u":[{"n":0, "v": "16110b00000040000003ef0000003d010000005e"}]}},
           {"v": 4, "r": 1, "0": {"u":[{"n":0, "v": "16110b0000027700000a6200000259010000008d"}]}}]
}
```
[Oct 01 114557 HAA Log.log](https://github.com/RavenSystem/esp-homekit-devices/files/12776862/Oct.01.114557.HAA.Log.log)

RavenSystem, 2023-10-02:
It is a bug caused by a duplicated line of code. It will be fixed with next version.

FYI: Command `16110b000000070000036b0000001202000003a` is wrong. It has 39 characters, and must be even. `a` will be ignored.

Tellicious, 2023-10-02:
You're right. Wrong copy/pase from my side

Tellicious, 2023-10-05:
This looks solved in 12.9.0b1

closed 2023-10-06 by RavenSystem