RavenSystem/esp-homekit-devices #2146: Temperature and Humidity POST

adamgdynia, 2023-11-02:
Hi! I have been using DHT22 temp&hum sensor for years and posting the data to thingsboard to make dashboards. Right now I'm preparing a set for a new location and I have noticed that there are interruptions in sending the data. Below is my code for Shelly2 and a screenshot of the dashboard. Notice how the green and blue line ends. Other readings are from different system.

I have red somewhere here that HAA is sensing the change in temp and hum values and posting them when they change. Is that true?

`{
  "c": {
    "io": [ [[4],2] ],
    "b": [ [ 0, 5 ] ]
  },
  "a": [{
    "t": 20,
    "0": { 
        "r": [ [ 4 ] ],
        "h": [{
          "h": "",
          "p": 8080,
          "m": 2,
          "e": "Content-type: application/json\r\n",
          "u": "api/v1//telemetry",
          "c": "{\"\":\"0\"}"
        }]
    },
    "1": { 
        "r": [ [ 4, 1 ] ],
        "h": [{
          "h": "",
          "p": 8080,
          "m": 2,
          "e": "Content-type: application/json\r\n",
          "u": "api/v1//telemetry",
          "c": "{\"\":\"1\"}"
        }]
       },
    "b": [ [ 0 ] ],
    "s": 0,
    "w": 1,
    "d": 600
  },
  {
    "t": 24,
    "g": 3,
    "n": 2,
    "j": 30,
    "z": 0,
    "y0": [{
      "v": 0,
      "r": 1,
      "0": {
          "h": [{
              "h": "",
              "p": 8080,
              "m": 2,
              "e": "Content-type: application/json\r\n",
              "u": "api/v1//telemetry",
              "c": "{\"\":#HAA@0000,\"\":#HAA@0001}"
      }]
      }
}]
  }]
}`

[image]

RavenSystem, 2023-11-24:
What HAA version are you using?

adamgdynia, 2023-11-25:
[image]

RavenSystem, 2023-11-25:
This should send temp and hum every 30 seconds, with or without changes. Logs are needed to know what is happening. You can enable buffered network logs with "o":8.

Other thing: you are using GPIO 0 as a button, but you forgot declare it into "io" array with [[0],6]

closed 2023-12-11 by RavenSystem