RavenSystem/esp-homekit-devices #2147: Shelly pro 1 v.1 support

lorenzomorandini, 2023-11-12:
Hi, I hope this is the right section. The wiki contains a config example for Shelly PRO 1 but it doesn't seem to work for the v.1. The relay actually never gets triggered.

https://kb.shelly.cloud/knowledge-base/shelly-pro-1-v1

Would it be possible to have a working config for it?

Thanks!

RavenSystem, 2023-11-12:
Try with this script and report:

`{"c":{"mc":[[113,14,4,8,8]],"io":[[[35],6,1,1],[[38,39],6,0,1],[[36],10,0,3]],"l":105,"b":[[35,5]]},"a":[{"0":{"r":[[107],[106]]},"1":{"r":[[107,1],[106,1]]},"b":[[38],[38,0],[39],[35]],"es":[{"t":22,"h":2,"n":5,"g":36,"j":10,"y0":[{"v":75,"r":1,"0":{"m":[[1]]}}]}]}]}`

RavenSystem, 2023-11-12:
Or try this. I think this is the right:

`{"c":{"mc":[[114,13,4,8,8]],"io":[[[35],6,1,1],[[38,39],6,0,1],[[36],10,0,3]],"l":102,"b":[[35,5]]},"a":[{"0":{"r":[[100],[101]]},"1":{"r":[[100,1],[101,1]]},"b":[[38],[38,0],[39],[35]],"es":[{"t":22,"h":2,"n":5,"g":36,"j":10,"y0":[{"v":75,"r":1,"0":{"m":[[1]]}}]}]}]}`

lorenzomorandini, 2023-11-12:
I tried both but with no luck. The first doesn't seem to change anything (the relay doesn't get triggered), the second doesn't even allow the switch to stay on in homekit. I checked the "reset homekit" flag, rebooted, readded the device to homekit.

RavenSystem, 2023-11-12:
Try this other:

`{"c":{"mc":[[114,13,4,8,8]],"io":[[[35],6,1],[[38,39],6,0,1],[[36],10,0,3]],"l":102,"b":[[35,5]]},"a":[{"0":{"r":[[100],[101]]},"1":{"r":[[100,1],[101,1]]},"b":[[38],[38,0],[39],[35]],"es":[{"t":22,"h":2,"n":5,"g":36,"j":10,"y0":[{"v":75,"r":1,"0":{"m":[[1]]}}]}]}]}`

lorenzomorandini, 2023-11-12:
Now the switch stays on again but still the relay doesn't get triggered. Just one thing: right now I can not connect it to a bulb or anything, so when I say the relay isn't triggered I mean I can not hear it getting triggered. Is it correct? When I got the device with the shelly firmware at first I could clearly hear the noise of the relay.

RavenSystem, 2023-11-12:
Forget relay. Are the LEDs doing something when you turn on/off HomeKit switch?

lorenzomorandini, 2023-11-12:
Nothing at all

RavenSystem, 2023-11-12:
Try this:

`{"c":{"mc":[[114,13,4,8,8]],"io":[[[35],6,1],[[38,39],6,0,1],[[36],10,0,3]],"l":105,"b":[[35,5]]},"a":[{"0":{"r":[[107],[106]]},"1":{"r":[[107,1],[106,1]]},"b":[[38],[38,0],[39],[35]],"es":[{"t":22,"h":2,"n":5,"g":36,"j":10,"y0":[{"v":75,"r":1,"0":{"m":[[1]]}}]}]}]}`

lorenzomorandini, 2023-11-12:
Works. The "out" led turns on and off and I can hear the relay noise. Thanks!

RavenSystem, 2023-11-12:
I updated script from Wiki. Thanks!

closed 2023-11-12 by RavenSystem