RavenSystem/esp-homekit-devices #2217: Button 1 doesn’t start any action on boot (until button 2 is toggled)

ferbulous, 2024-02-18:
**Detailed steps to reproduce it**
On boot, i need to press button 2 first so button 1 can trigger relay 1 and execute hold action. Otherwise it doesn’t execute any action

**MEPLHAA Script**

{"c":{"io":[[[5,15,13],2],[[16,12],6]],"l":13,"i":0,"b":[[16,5]]},"a":[{"0":{"r":[[5]]},"1":{"r":[[5,1]]},"b":[[16,1]],"es":[{"0":{"r":[[15]]},"1":{"r":[[15,1]]},"b":[[12,1]]}]}]}

**Used device**
2 gang Tuya switch

**HAA version**
12.11.4

**iOS/iPadOS version**
17.1
**Logs**
A .txt file with logs. Read [how to capture logs from your device](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs)

**Screen-shots or videos showing issue**

https://github.com/RavenSystem/esp-homekit-devices/assets/24526737/b737b5e8-aef8-4a95-9de1-deb613ec57f3

ferbulous, 2024-02-27:
Updated with logs

gd123294813, 2024-03-07:
问题解决了吗？？

ferbulous, 2024-03-07:
> 问题解决了吗？？

No, still not solved