RavenSystem/esp-homekit-devices #2218: Sonoff SV isolated mode type garage door begin unresponsive

pilot1981, 2024-02-20:
**Detailed steps to reproduce it**
Hi, random Apple Home say this device type garage door is unresponsive, device lost WIFI connection.

**MEPLHAA Script**
{"c":{"l":13,"b":[{"g":0,"t":5}],"io":[[[12,13],2],[[0],6],[[4,14],6,1]],"n":"nuovocancello","tz":"UTC+1,M3.4.0/2,M10.5.0/3","ot":"[192.168.6.41:514](http://192.168.6.41:514/)","o":8,"ntp":"[192.168.6.41](http://192.168.6.41/)","0":{"r":[{"g":12}]}},"a":[{"0":{},"1":{"r":[{"g":12,"v":1,"i":0.5}]},"2":{},"3":{"r":[{"g":12,"v":1,"i":0.5}]},"4":{"m":[[2,1],[2,-1]]},"t":40,"d":18,"b":[{"g":0}]},{"0":{"m":[[1,11]]},"xa":0,"t":1,"h":0,"b":[],"s":0,"d":60},{"t":1,"d":1}]}

**Used device**
SONOFF SV isolated mode

**HAA version**
12.11.4

**iOS/iPadOS version**
iOS 17.3.1

**Logs**
[logs HAA.txt](https://github.com/RavenSystem/esp-homekit-devices/files/14341171/logs.HAA.txt)

**Screen-shots or videos showing issue**
NA

pilot1981, 2024-02-20:
today, another stop of device on Apple Home, but WIFi connection working...

I attach new logs...I don0t found any useful informations
[logs HAA2.txt](https://github.com/RavenSystem/esp-homekit-devices/files/14345116/logs.HAA2.txt)

pilot1981, 2024-03-07:
problem still here...

RavenSystem, 2024-03-08:
It is not a HAA firmware bug. Try improving WiFi signal adding an extra AP or upgrading your WiFi system.

closed 2024-03-08 by RavenSystem

pilot1981, 2024-03-08:
You see it form logs? Where?

I have another sonoff SV 5 cm far…