RavenSystem/esp-homekit-devices #2234: Script MELPHAA tourne en boucle

MrVince06, 2024-03-20:
**J’ai suivi les étapes pour arriver au niveau de la page jaune. Le Shelly est alors détectable avec les wifi HAAxxxxxx. J’ai rentré le script + le wifi. La page m’affiche ok. Après plus d’un heure, le script s’exécute toujours sur l’appli HAA. Le wifi HAAxxxx n’est plus disponible donc impossible de retourner au menu avec la page jaune.**

**{"c":{"io":[[[12,13,0],2],[[4],6,1],[[5,18],6,0,1],[[35],10,0,3],[[27]]],"l":0,"b":[[4,5]],"ic":[[25,26,100]]},"a":[{"t":45,"o":29,"c":27,"f":70,"0":{"r":[[13],[12,1]],"m":[[2,-10001],[3,-10001]]},"1":{"r":[[13,1],[12]],"m":[[2,-10001],[3,-10001]]},"2":{"r":[[13],[12]],"m":[[2,-10000],[3,-10000]]},"3":{"r":[[13],[12,0,0.2]],"m":[[2,-10001],[3,-10001]]},"4":{"r":[[13,0,0.2],[12]],"m":[[2,-10001],[3,-10001]]},"8":{"m":[[0,101]]},"f0":[[5,0]],"f1":[[18,0]],"f2":[[5],[18]],"es":[{"t":75,"h":2,"ks":0,"j":2,"cd":0.95,"n":2,"dt":[0,56],"vf":0.0000382602,"vo":-0.068,"cf":0.00000949523,"co":-0.017,"pf":0.006097560976,"y1":[{"v":2,"r":1,"0":{"m":[[1,-1]]}},{"v":2.5}]},{"t":75,"h":2,"ks":0,"j":2,"cd":0.45,"n":3,"dt":[0,56],"vf":0.0000382602,"vo":-0.068,"cf":0.00000949523,"co":-0.017,"pf":0.006097560976,"y1":[{"v":2,"r":1,"0":{"m":[[1,-1]]}},{"v":2.5}]},{"t":22,"h":2,"n":5,"g":35,"j":5,"y0":[{"v":75,"r":1,"0":{"m":[[1,-1]]}}]}]}]}**

**Shelly +1PM**

**7.10.3**

**iOS 17.4**

**-PJAX-Container, Turbo-Visit, Turbo-Frame, Accept-Encoding, Accept, X-Requested-With
0024.805 192.168.0.133-59C3F4: Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/166993980/c1f86db7-9a6a-4cef-a5c8-c4fafe674b74?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240320%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240320T173516Z&X-Amz-Expires=300&X-Amz-Signature=adb8740e3b6a6dad3be3b0117dcb8ab4377f7784983907177d1a6a10ceef0d33&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain_esp32_1.bin&response-content-type=application%2Foctet-stream
0024.851 192.168.0.133-59C3F4: Cache-Control: no-cache
0024.863 192.168.0.133-59C3F4: Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
0024.864 192.168.0.133-59C3F4: X-Frame-Options: deny
0024.864 192.168.0.133-59C3F4: X-Content-Type-Options: nosniff
0024.864 192.168.0.133-59C3F4: X-XSS-Protection: 0
0024.875 192.168.0.133-59C3F4: Referrer-Policy: no-referrer-when-downgrade
0024.875 192.168.0.133-59C3F4: Content-Security-Policy: default-src 'none'; base-uri 'self'; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com www.githubstatus.com collector.github.com raw.githubusercontent.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com api.githubcopilot.com objects-origin.githubusercontent.com *.actions.githubusercontent.com wss://*.actions.githubusercontent.com productionresultssa0.blob.core.windows.net/ productionresultssa1.blob.core.windows.net/ productionresultssa2.blob.core.windows.net/ productionresultssa3.blob.core.windows.net/ productionresultssa4.blob.core.windows.net/ productionresultssa5.blob.core.windows.net/ productionresultssa6.blob.core.windows.net/ productionresultssa7.blob.core.windows.net/ productionresultssa8.blob.core.windows.net/ productionresultssa9.blob.core.windows.neHTTP returns 302

0024.967 192.168.0.133-59C3F4: 
0024.969 192.168.0.133-59C3F4: *** FORWARDING objects.githubusercontent.com:443/github-production-release-asset-2e65be/166993980/c1f86db7-9a6a-4cef-a5c8-c4fafe674b74?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240320%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240320T173516Z&X-Amz-Expires=300&X-Amz-Signature=adb8740e3b6a6dad3be3b0117dcb8ab4377f7784983907177d1a6a10ceef0d33&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain_esp32_1.bin&response-content-type=application%2Foctet-stream
0025.016 192.168.0.133-59C3F4: *** CONNECT
0025.016 192.168.0.133-59C3F4: DNS..OK
0025.017 192.168.0.133-59C3F4: Socket..OK
0025.017 192.168.0.133-59C3F4: Connect..OK
0025.054 192.168.0.133-59C3F4: SSL..OK
0025.055 192.168.0.133-59C3F4: FD..OK
0025.056 192.168.0.133-59C3F4: SNI..OK
0025.057 192.168.0.133-59C3F4: SSLConnect..OK
0025.783 192.168.0.133-59C3F4: Write..OK
0028.152 192.168.0.133-59C3F4: GOT 866 Bytes
0028.153 192.168.0.133-59C3F4: HTTP/1.1 206 Partial Content
0028.153 192.168.0.133-59C3F4: Connection: keep-alive
0028.153 192.168.0.133-59C3F4: Content-Length: 2
0028.153 192.168.0.133-59C3F4: Content-Type: application/octet-stream
0028.154 192.168.0.133-59C3F4: Content-MD5: guLSoQ/3Nl+SFUyF6pU7Pw==
0028.164 192.168.0.133-59C3F4: Last-Modified: Fri, 08 Mar 2024 14:44:18 GMT
0028.165 192.168.0.133-59C3F4: ETag: "0x8DC3F7E3BD27DF8"
0028.165 192.168.0.133-59C3F4: Server: Windows-Azure-Blob/1.0 Microsoft-HTTPAPI/2.0
0028.176 192.168.0.133-59C3F4: x-ms-request-id: ae7efb06-101e-0044-3667-716b6a000000
0028.177 192.168.0.133-59C3F4: x-ms-version: 2020-10-02
0028.177 192.168.0.133-59C3F4: x-ms-creation-time: Fri, 08 Mar 2024 14:44:18 GMT
0028.188 192.168.0.133-59C3F4: x-ms-lease-status: unlocked
0028.188 192.168.0.133-59C3F4: x-ms-lease-state: available
0028.188 192.168.0.133-59C3F4: x-ms-blob-type: BlockBlob
0028.189 192.168.0.133-59C3F4: Content-Disposition: attachment; filename=otamain_esp32_1.bin
0028.200 192.168.0.133-59C3F4: x-ms-server-encrypted: true
0028.200 192.168.0.133-59C3F4: Via: 1.1 varnish, 1.1 varnish
0028.200 192.168.0.133-59C3F4: Accept-Ranges: bytes
0028.201 192.168.0.133-59C3F4: Age: 2885
0028.201 192.168.0.133-59C3F4: Content-Range: bytes%2F20240320%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240320T173516Z&X-Amz-Expires=300&X-Amz-Signature=adb8740e3b6a6dad3be3b0117dcb8ab4377f7784983907177d1a6a10ceef0d33&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dotamain_esp32_1.bin&response-content-type=application%2Foctet-stream

0028.276 192.168.0.133-59C3F4: 
0028.276 192.168.0.133-59C3F4: *** CONNECT
0028.277 192.168.0.133-59C3F4: DNS..OK
0028.277 192.168.0.133-59C3F4: Socket..OK
0028.277 192.168.0.133-59C3F4: Connect..OK
0028.297 192.168.0.133-59C3F4: SSL..OK
0028.298 192.168.0.133-59C3F4: FD..OK
0028.298 192.168.0.133-59C3F4: SNI..OK
0028.299 192.168.0.133-59C3F4: SSLConnect..OK
0029.866 192.168.0.133-59C3F4: - 4096
0030.706 192.168.0.133-59C3F4: - 8192
0031.556 192.168.0.133-59C3F4: - 12288
0032.386 192.168.0.133-59C3F4: - 16384
0033.236 192.168.0.133-59C3F4: - 20480
0034.066 192.168.0.133-59C3F4: - 24576
0034.896 192.168.0.133-59C3F4: - 28672
0035.726 192.168.0.133-59C3F4: - 32768
[... 375 lines cut ...]
0036.757 192.168.0.133-59C3F4: - 40960
0037.316 192.168.0.133-59C3F4: - 45056
0038.156 192.168.0.133-59C3F4: - 49152
0038.986 192.168.0.133-59C3F4: - 53248
0039.856 192.168.0.133-59C3F4: - 57344
0048.406 192.168.0.133-59C3F4: - 61440
0049.276 192.168.0.133-59C3F4: - 65536
0049.866 192.168.0.133-59C3F4: - 69632
0050.726 192.168.0.133-59C3F4: - 73728
0051.556 192.168.0.133-59C3F4: - 77824
0052.136 192.168.0.133-59C3F4: - 81920
0053.006 192.168.0.133-59C3F4: - 86016
0053.836 192.168.0.133-59C3F4: - 90112
0054.706 192.168.0.133-59C3F4: - 94208
0055.556 192.168.0.133-59C3F4: - 98304
0057.246 192.168.0.133-59C3F4: - 106496
0058.976 192.168.0.133-59C3F4: - 114688
0059.846 192.168.0.133-59C3F4: - 118784
0061.556 192.168.0.133-59C3F4: - 126976
0062.416 192.168.0.133-59C3F4: - 131072
[Mar 20 183907 HAA Log.log](https://github.com/RavenSystem/esp-homekit-devices/files/14670523/Mar.20.183907.HAA.Log.log)
**
A .txt file with logs. Read [how to capture logs from your device](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs)

**Screen-shots or videos showing issue**

Jdas33, 2024-03-28:
I have a same problem with Shelly plus 2 PM.
An infinite loop appears.

HAA Installer 7.10.3

I can't connect with the browser. How to resolve the situation?

RavenSystem, 2024-04-02:
Bad OTA Data partition. Erase and flash again.

closed 2024-04-02 by RavenSystem