RavenSystem/esp-homekit-devices #2237: Install loop

miguelpruivo, 2024-04-02:
My brand new Shelly Plus 2PM is stuck in an install loop. It installs and then verifies and then installs and then verifies and so on so fourth.

here’s the logs:
```
0105.585 10.0.2.174-A61820: - 503808
0106.215 10.0.2.174-A61820: - 507904
0106.815 10.0.2.174-A61820: - 512000
0107.425 10.0.2.174-A61820: - 516096
0108.035 10.0.2.174-A61820: - 520192
0108.625 10.0.2.174-A61820: - 524288
0109.235 10.0.2.174-A61820: - 528384
0109.865 10.0.2.174-A61820: - 532480
0111.745 10.0.2.174-A61820: - 544768
0112.395 10.0.2.174-A61820: - 548864
0112.995 10.0.2.174-A61820: - 552960
0114.214 10.0.2.174-A61820: - 561152
0114.825 10.0.2.174-A61820: - 565248
0115.435 10.0.2.174-A61820: - 569344
0116.045 10.0.2.174-A61820: - 573440
0116.654 10.0.2.174-A61820: - 577536
0117.285 10.0.2.174-A61820: - 581632
0117.545 10.0.2.174-A61820: - 585728
0118.136 10.0.2.174-A61820: - 589824
0118.725 10.0.2.174-A61820: - 593920
0119.366 10.0.2.174-A61820: - 598016
0120.544 10.0.2.174-A61820: - 606208
0121.144 10.0.2.174-A61820: - 610304
0121.735 10.0.2.174-A61820: - 614400
0122.375 10.0.2.174-A61820: - 618496
0123.005 10.0.2.174-A61820: - 622592
0123.635 10.0.2.174-A61820: - 626688
0124.225 10.0.2.174-A61820: - 630784
0124.824 10.0.2.174-A61820: - 634880
0125.435 10.0.2.174-A61820: - 638976
0126.025 10.0.2.174-A61820: - 643072
0126.615 10.0.2.174-A61820: - 647168
0128.075 10.0.2.174-A61820: - 659456
0128.654 10.0.2.174-A61820: - 663552
0129.225 10.0.2.174-A61820: - 667648
0129.806 10.0.2.174-A61820: - 671744
0130.382 10.0.2.174-A61820: - 675104
0130.386 10.0.2.174-A61820: >>> Verifying

Adv Log (c) 2022-2024 José A. Jiménez Campos

0011.293 10.0.2.174-A61820: >>> Get sign
0011.294 10.0.2.174-A61820: *** DOWNLOADING
0011.295 10.0.2.174-A61820: *** FORWARDING github.com:443/RavenSystem/haa/releases/latest/download/otamain_esp32.bin.sec
0011.296 10.0.2.174-A61820: *** CONNECT
0011.296 10.0.2.174-A61820: DNS..OK
0011.408 10.0.2.174-A61820: Socket..OK
0011.409 10.0.2.174-A61820: Connect..OK
0011.465 10.0.2.174-A61820: SSL..OK
0011.467 10.0.2.174-A61820: FD..OK
0011.467 10.0.2.174-A61820: SNI..OK
0011.467 10.0.2.174-A61820: SSLConnect..OK
0012.470 10.0.2.174-A61820: Write..OK
ain_esp32.bin.sec
0014.719 10.0.2.174-A61820: Cache-Control: no-cache
0014.720 10.0.2.174-A61820: Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
0014.721 10.0.2.174-A61820: X-Frame-Options: deny
0014.722 10.0.2.174-A61820: X-Content-Type-Options: nosniff
0014.732 10.0.2.174-A61820: X-XSS-Protection: 0
0014.733 10.0.2.174-A61820: Referrer-Policy: no-referrer-when-downgrade
0014.734 10.0.2.174-A61820: Content-Security-Policy: default-src 'none'; base-uri 'self'; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com www.githubstatus.com collector.github.com raw.githubusercontent.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com githOK
0014.926 10.0.2.174-A61820: SSL..OK
0014.927 10.0.2.174-A61820: FD..OK
0014.927 10.0.2.174-A61820: SNI..OK
0014.928 10.0.2.174-A61820: SSLConnect..OK
0015.944 10.0.2.174-A61820: Write..OK
```

RavenSystem, 2024-04-02:
Use last version. Outdated versions are not longer supported.

closed 2024-04-02 by RavenSystem

miguelpruivo, 2024-04-02:
@RavenSystem how can I use the last version if the Shelly is trying to update on its own to 7.3.2 I believe and it can't update? How am I supposed to update then?

RavenSystem, 2024-04-02:
You said that it is a new installation, then HAA is trying to install, not update. You must flash lastest version of fullhaaboot file to avoid issues. Firmware running is haaboot, not otamain.