RavenSystem/esp-homekit-devices #2240: Sonoff S20 With DHT22

Gandalf8266, 2024-04-03:
**Detailed steps to reproduce it**
There must be a Problem wit HAA 12.12.4, I tried a third S20 with a DH22, I made a clean install, configured the DHT22 at GPIO3, attendet it at homekit. The Socket works with homekit, as soon as I connect the DHT22 at GPIO3 (I also testet GPIO1) the Socket works only one time, then it doesn't answer. As soon as I disconnect the DHT22 an disconnect/connect the S20 to power again, it works again. I made a screenshotmovie what happens when I has connected the DHT22 and I'm swichting in home-App.

**MEPLHAA Script**
{"c":{"io":[[[12,13],2],[[0],6]],"l":13,"b":[[0,5]]},"a":[{"t":24,"g":1,"n":2},{"t":2,"0":{"r":[[12]]},"1":{"r": [[12,1]]},"b":[[0]]}]}
**Used device**
Sonoff S20
**HAA version**
12.12.4
**iOS/iPadOS version**
17.4.1
**Logs**
A .txt file with logs. Read [how to capture logs from your device](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs)
No log so far.
**Screen-shots or videos showing issue**
https://share.icloud.com/photos/068AL5PtoDFEdMQa4MiBRDFrQ

RavenSystem, 2024-04-19:
I can not reproduce the issue.

RavenSystem, 2024-04-19:
Could you try with version 12.11.4? Read how to install a specific version:
https://github.com/RavenSystem/esp-homekit-devices/wiki/installation#installing-a-specific-version

Gandalf8266, 2024-04-28:
hi, I have Problems to add the devices to home app again.
I did a clean install with the version 12.11.4 as following:

_christian@MacBook-Pro-von-Christian-8 8266-Bausteine % python3 -m esptool -p /dev/tty.usbserial-A600dJRY erase_flash                                                                                                 
esptool.py v4.7.0
Serial port /dev/tty.usbserial-A600dJRY
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: dc:4f:22:2d:84:81
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.5s
Hard resetting via RTS pin...
christian@MacBook-Pro-von-Christian-8 8266-Bausteine % python3 -m esptool -p /dev/tty.usbserial-A600dJRY -b 115200 --before=default_reset --after=hard_reset write_flash -fs 1MB -fm dout 0x0 fullhaaboot-12.11.4.bin
esptool.py v4.7.0
Serial port /dev/tty.usbserial-A600dJRY
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: dc:4f:22:2d:84:81
Stub is already running. No upload is necessary.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00062fff...
Compressed 402292 bytes to 286032...
Wrote 402292 bytes (286032 compressed) at 0x00000000 in 25.4 seconds (effective 126.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
christian@MacBook-Pro-von-Christian-8 8266-Bausteine %_ 

After connecting the socket to power (socket in the wall) I'm connecting the web interface with http://192.168.4.1:4567/, I configured WIFI and added the MEPLHAA Script:
{"c":{"io":[[[12,13],2],[[0],6]],"l":13,"b":[[0,5]]}," a":[{"t":22,"g":3,"n":3},{"t":2,"0":{"r":[[12]]},"1":{"r": [[12,1]]},"b":[[0]]}]}
pressing ok, the "ok" appears in the browser.
 I'm waiting at least 10 minutes or longer, nothings happens. I'm not able to add it on HomeKit. 
No LED ist flashing, the button on the S20 Socket doesn't work.

I think the sockets are broken. Because sometimes I was not able to flash, the process interrupted suddenly. I tried it again, same problem. I waited one day, than it works again with same commands, very strange to me.

Gandalf8266, 2024-04-28:
After a clean install, there is no HomeKit reset possibile, an in my opinion not necessary 

[image]

RavenSystem, 2024-05-09:
Issue is not reproducible. It seems a faulty hardware, related to flash storage or power supply.

closed 2024-05-09 by RavenSystem