RavenSystem/esp-homekit-devices #2241: Garage door with Shelly1 small issues

iondarie, 2024-04-10:
Hello,

I have a garage door with 2 sensors connected to a Shelly1 v3. The open and close sensors are connected to the RX/TX GPIO's. I have setup the open/close timer to 45s (the door takes about 41-43s to open/close).

I am having some issues with this setup, sometimes i get "Obstructed" status in homekit even if the door is fully open/closed. In the past week a few times the status in homekit showed Open but the door was closed, also notifications are not working sometimes. 

Anyway here is my code (tried it with 2 shelly's so i can be sure it's not a hardware issue hopefully):

```
{
  "c":{
    "io":[[[1,3],6,1],[[4],2]]
  },
  "a":[{
    "t":40,
    "s":5,
    "d":45,
    "c":45,
    "e":2,
    "vs":1,
    "0":{"r":[{"g":4,"v":1,"i":0.5}]},
    "1":{"a":0},
    "2":{"r":[[4,0,1.5],[4,1,0.5],[4,1,2]]},
    "3":{"a":2},
    "10":{"a":0},
    "12":{"a":0},
    "13":{"a":0},
    "f2":[[1,0]],
    "f3":[[3,0]],
    "f4":[[3]],
    "f5":[[1]]
  }]
}
```

Please advise if i have done something wrong. 
Thank you for your time

mariusvasss, 2024-04-10:
Salut. 

Nu mai folosi TX-ul, foloseste alt GPIO daca mai ai.

iondarie, 2024-04-11:
Salut, din pacate singurul GPIO safe ramas este GPIO5 (butonul SW) dar nu vrea sa citeasca senzorul daca il pun pe acest gpio...

mariusvasss, 2024-04-11:
Atunci lasa doar senzorul de inchis si ai rezolvat problema. Eu folosesc doar un senzor pentru inchis si este minunat.