RavenSystem/esp-homekit-devices #2245: MCP23017 with ESP8266 D1 mini

duku602, 2024-04-23:
I am trying to use MCP 23017 expansion board, the I2C is working, I can access the A channel as input but B channel is not working.

Again, not matter what I do, the output (channel B in my case) will not work, I even tried the same code (an lots of variations including inverted logic) with ANOTHER mcp dip and same results.

PLEASE, someone to make a instructable on how to proper use expansion boards with HAA ! ! ! (it will be a lot easier to use them with sonoffs instead of esp32 + 220v to 5v module or LDOs.

**Detailed steps to reproduce it**
A0 + A1 + A2 TO GND, address 0x20
[MCP23017] SCL to D1 (gpio5)
[MCP23017] SDA to D2 (gpio4)
[MCP23017] GPB 7 (pin8) [haa pin 115] to relay (trigger active high)  --- THIs channel is not working
[MCP23017] GPA 0 (pin21) [haa pin 100] to touch module - is working properly

PLEASE HELP!!! I've spent days already searching on internet for someone that used HAA with MCP 23017.

Thank you !

**MEPLHAA Script**
{
  "c": {
    "c": 1,
    "ic" : [ [ 5, 4, 80, 1, 1] ],
    "mc" : [ [ 0, 32, 258, 0 ] ],
    "io": [ 
        [[ 100], 6, 0,1],
        [[108, 115], 2]
    ],
    "b": [ [ 0, 5 ] ],
    "u": 1, "m":30,"sn":"AtticBathroom"
  },
  "a": [
      {
        "t": 1,
        "s": 5,
        "0": { "r": [ [ 115, 0 ] ] },
        "1": { "r": [ [ 115, 1 ] ] },
        "b": [ [ 100 ] ]
      }
  ]
}

**Used device**
esp 8266 D1 mini + mcp23017 [dip]

**HAA version**
12.12.4

**iOS/iPadOS version**
latest

**Logs**
A .txt file with logs. Read [how to capture logs from your device](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs)

Adv Log (c) 2022-2024 José A. Jiménez Campos

Jan 01 00:00:25 192.168.1.222-A59780 Free Heap 46496
Jan 01 00:00:25 192.168.1.222-A59780 Start HK
Jan 01 00:00:25 192.168.1.222-A59780 HK ID: AF:2B:55:40:D1:FF
Jan 01 00:00:25 192.168.1.222-A59780 Found B5706974-F748-4570-95E2-D3EA8DC1FFD1
Jan 01 00:00:25 192.168.1.222-A59780 Free Heap 38024

Adv Log (c) 2022-2024 José A. Jiménez Campos

Jan 01 00:00:25 192.168.1.222-A59780 Free Heap 46496
Jan 01 00:00:25 192.168.1.222-A59780 Start HK
Jan 01 00:00:25 192.168.1.222-A59780 HK ID: AF:2B:55:40:D1:FF
Jan 01 00:00:25 192.168.1.222-A59780 Found B5706974-F748-4570-95E2-D3EA8DC1FFD1
Jan 01 00:00:25 192.168.1.222-A59780 Free Heap 38024
Jan 01 00:00:26 192.168.1.222-A59780 mDNS prob 1
Jan 01 00:00:26 192.168.1.222-A59780 mDNS prob 1
Jan 01 00:00:26 192.168.1.222-A59780 mDNS Name=HAA-A59780 md=RavenSystem HAA Merlinpv=1.1id=AF:2B:55:40:D1:FFc#=19s#=25sf=0ci=8
                                                                                                                               sh=IHQW8w== Port=5556 TTL=4500
Jan 01 00:00:26 192.168.1.222-A59780 <0> Run A2
Jan 01 00:00:26 192.168.1.222-A59780 mDNS Name=HAA-A59780 md=RavenSystem HAA Merlinpv=1.1id=AF:2B:55:40:D1:FFc#=19s#=25sf=0ci=8
                                                                                                                               sh=IHQW8w== Port=5556 TTL=4500Jan 01 00:00:26 192.168.1.222-A59780 [6] New 192.168.1.30:2509 1/12 HEAP 35500
Jan 01 00:00:26 192.168.1.222-A59780 [6] Verify 1/2
Jan 01 00:00:26 192.168.1.222-A59780 [6] New 192.168.1.30:2509 1/12 HEAP 35500
Jan 01 00:00:26 192.168.1.222-A59780 [6] Verify 1/2
Jan 01 00:00:26 192.168.1.222-A59780 [6] Verify 2/2
Jan 01 00:00:26 192.168.1.222-A59780 [6] Verify 2/2
Jan 01 00:00:26 192.168.1.222-A59780 [6] Verify B5706974-F748-4570-95E2-D3EA8DC1FFD1
Jan 01 00:00:26 192.168.1.222-A59780 [6] Verify B5706974-F748-4570-95E2-D3EA8DC1FFD1
Jan 01 00:00:26 192.168.1.222-A59780 [6] Verify OK
Jan 01 00:00:26 192.168.1.222-A59780 [6] Verify OK
Jan 01 00:00:26 192.168.1.222-A59780 [6] Get ACC
Jan 01 00:00:27 192.168.1.222-A59780 [6] Get CH
Jan 01 00:00:26 192.168.1.222-A59780 [6] Get ACC
Jan 01 00:00:27 192.168.1.222-A59780 [6] Get CH
Jan 01 00:00:28 192.168.1.222-A59780 NTP 192.168.1.1 (-1)
Jan 01 00:00:28 192.168.1.222-A59780 NTP 192.168.1.1 (-1)
Jan 01 00:00:33 192.168.1.222-A59780 mDNS prob 2
Jan 01 00:00:33 192.168.1.222-A59780 NTP 192.168.1.1 (-1)
Jan 01 00:00:33 192.168.1.222-A59780 mDNS prob 2
Jan 01 00:00:33 192.168.1.222-A59780 NTP 192.168.1.1 (-1)
Apr 23 16:41:14 192.168.1.222-A59780 NTP pool.ntp.org (0)
Apr 23 16:41:14 192.168.1.222-A59780 NTP pool.ntp.org (0)
Apr 23 16:41:17 192.168.1.222-A59780 mDNS TTL 4500/2179s
Apr 23 16:41:17 192.168.1.222-A59780 mDNS TTL 4500/2179s
Apr 23 16:41:28 192.168.1.222-A59780 [6] Upd CH
Apr 23 16:41:28 192.168.1.222-A59780 [6] Upd CH
Apr 23 16:41:28 192.168.1.222-A59780 [6] Get CH
Apr 23 16:41:28 192.168.1.222-A59780 [6] Get CH
Apr 23 16:41:29 192.168.1.222-A59780 [6] Get CH
Apr 23 16:41:29 192.168.1.222-A59780 [6] Get CH
Apr 23 16:41:29 192.168.1.222-A59780 [6] Get CH
Apr 23 16:41:29 192.168.1.222-A59780 [6] Get CH
Apr 23 16:41:29 192.168.1.222-A59780 [6] Get CH
Apr 23 16:41:30 192.168.1.222-A59780 [6] Upd CH
Apr 23 16:41:30 192.168.1.222-A59780 [6] Upd CH
Apr 23 16:41:30 192.168.1.222-A59780 <1> -> SW ON 1
Apr 23 16:41:30 192.168.1.222-A59780 Setup 1/8
Apr 23 16:41:30 192.168.1.222-A59780 <1> Run A1
Apr 23 16:41:30 192.168.1.222-A59780 <1> DigO 115->1 (0)
Apr 23 16:41:30 192.168.1.222-A59780 [6] Send Ev
Apr 23 16:41:30 192.168.1.222-A59780 <1> -> SW ON 1
Apr 23 16:41:30 192.168.1.222-A59780 Setup 1/8
Apr 23 16:41:30 192.168.1.222-A59780 <1> Run A1
Apr 23 16:41:30 192.168.1.222-A59780 <1> DigO 115->1 (0)
Apr 23 16:41:30 192.168.1.222-A59780 [6] Send Ev
Apr 23 16:41:33 192.168.1.222-A59780 Saving
Apr 23 16:41:33 192.168.1.222-A59780 Saving
Apr 23 16:41:35 192.168.1.222-A59780 [6] Upd CH
Apr 23 16:41:35 192.168.1.222-A59780 <1> -> SW ON 0
Apr 23 16:41:35 192.168.1.222-A59780 Setup 1/8
Apr 23 16:41:35 192.168.1.222-A59780 [6] Upd CH
Apr 23 16:41:35 192.168.1.222-A59780 <1> -> SW ON 0
Apr 23 16:41:35 192.168.1.222-A59780 Setup 1/8
Apr 23 16:41:35 192.168.1.222-A59780 <1> Run A0
Apr 23 16:41:35 192.168.1.222-A59780 <1> Run A0
Apr 23 16:41:38 192.168.1.222-A59780 Saving
Apr 23 16:41:38 192.168.1.222-A59780 Saving
Apr 23 16:42:05 192.168.1.222-A59780 [6] Upd CH
Apr 23 16:42:05 192.168.1.222-A59780 <1> -> SW ON 1
Apr 23 16:42:05 192.168.1.222-A59780 Setup 1/8
Apr 23 16:42:05 192.168.1.222-A59780 [6] Upd CH
Apr 23 16:42:05 192.168.1.222-A59780 <1> -> SW ON 1
Apr 23 16:42:05 192.168.1.222-A59780 Setup 1/8
Apr 23 16:42:05 192.168.1.222-A59780 <1> Run A1
Apr 23 16:42:05 192.168.1.222-A59780 <1> DigO 115->1 (0)
Apr 23 16:42:08 192.168.1.222-A59780 Saving
Apr 23 16:42:08 192.168.1.222-A59780 Saving
Apr 23 16:42:28 192.168.1.222-A59780 <1> DigI 2 GPIO 100
Apr 23 16:42:28 192.168.1.222-A59780 <1> -> SW ON 0
Apr 23 16:42:28 192.168.1.222-A59780 Setup 1/8
Apr 23 16:42:28 192.168.1.222-A59780 <1> Run A0
Apr 23 16:42:28 192.168.1.222-A59780 <1> DigI 2 GPIO 100
Apr 23 16:42:28 192.168.1.222-A59780 <1> -> SW ON 0
Apr 23 16:42:28 192.168.1.222-A59780 Setup 1/8
Apr 23 16:42:28 192.168.1.222-A59780 <1> Run A0
Apr 23 16:42:28 192.168.1.222-A59780 <1> DigO 115->0 (0)
Apr 23 16:42:28 192.168.1.222-A59780 <1> DigO 115->0 (0)
Apr 23 16:42:29 192.168.1.222-A59780 [6] Get CH
Apr 23 16:42:29 192.168.1.222-A59780 [6] Get CH

**Screen-shots or videos showing issue**

RavenSystem, 2024-04-25:
Try removing this part, because it is not needed since output is declared at MCP array, and I don't know if it can create issues: `[[108, 115], 2]`

RavenSystem, 2024-04-25:
There is a bug with channel B outputs. it will be fixed in next version. With current version, channel A works fine for outputs, then you can use it, and channel B for inputs.

duku602, 2024-04-25:
Hello again and thank you for the quick response.

I did tried (before opening the issue) to use Channel B as input and channel A as output but, again, channel B [input in that case] was not reliable, it was self triggering randomly.  Channel A worked like a charm as both INPUT or OUTPUTS; Channel B was failing as both input or output.

again, thank you for everything!

RavenSystem, 2024-04-25:
Use this script, with channel A as output and channel B as input. Connect your button to GND and Pin.
```json
{
"c": {
"c": 1,
"ic" : [ [ 5, 4, 80, 1, 1] ],
"mc" : [ [ 0, 32, 0, 256 ] ],
"io": [
[[ 0, 100 ], 6],
],
"b": [ [ 0, 5 ] ],
"u": 1, "m":30,"sn":"AtticBathroom"
},
"a": [
{
"t": 1,
"s": 5,
"0": { "r": [ [ 100, 0 ] ] },
"1": { "r": [ [ 100, 1 ] ] },
"b": [ [ 115 ] ]
}
]
}
```

duku602, 2024-04-25:
works perfectly!  Thank you!

RavenSystem, 2024-05-09:
Fixed with v12.12.5 https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_12.12.5

closed 2024-05-09 by RavenSystem