RavenSystem/esp-homekit-devices #2259: iFan04 lights turn on after wifi outage

blueyetisoftware-jason, 2024-05-10:
- Ensure the iFan04 if connected to the wifi and that the lights are off
- Cut the wifi and let it sit
- Reconnect the wifi
- Upon the fan reconnecting, the lights will flip on

Same things happens with a power outage, but I narrowed it down to wifi outages in particular.

**Used device**
Sonoff iFan04

**HAA version**
12.11.4

RavenSystem, 2024-05-20:
ESP8266 WiFi driver (closed source) is not fully compatible with some WiFi systems, causing a reboot. If you don't want that iFan turn on light at boot, remove `"s":1` from the script.

closed 2024-05-20 by RavenSystem

blueyetisoftware-jason, 2024-05-21:
@RavenSystem  Thanks for the tip. Will that leave it in its last state, or will it shut it off as opposed to on?

RavenSystem, 2024-05-21:
It will shut it off. If you want to keep last state, replace it with `"s":5`.

What is your brand and model WiFi system?