RavenSystem/esp-homekit-devices #2262: Device shows wrong temperature after updating from 12.12.4 --> 12.12.5

Gandalf8266, 2024-05-17:
I have a ESP8266 microcontroller with a DHT22 connected. 
Device works fine so far, but on every update it shows wrong temperature (-99°C) and humidity (0%).

It happens on every update, but now, I made screenshots bevor und after update. 
It's easy to fix, I only disconnect from power and connect again, but this is annoying.

The config is here:
{"c":{"o":0},"a":[{"t":24,"b":[],"n":2,"g":2}]}

I have attached screenshots bevor und after update. 
[image]
[image]
[image]
[image]

RavenSystem, 2024-05-20:
It is hardware related. GPIO 2 is pulled up and high at boot. This can not be changed by firmware. Better, use other GPIO.
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

closed 2024-05-20 by RavenSystem

vsidlu, 2024-09-06:
Strange, but sometimes it shows the correct reading, but sometimes -99. What could this be related to?

vsidlu, 2024-10-01:
I replaced all DHT22 temperature sensors. I no longer encounter this issue.