RavenSystem/esp-homekit-devices #2280: Installed update 12.12.8 using HAA Manager and devices stop working

rickdhoy, 2024-07-30:
**Detailed steps to reproduce it**

Installed new update using HAA Manager 12.12.8 on a number of Sonoff devices.

They were on 11.7.1 prior to the update.

The HAA manager version is 3.8.0

The iPad is on 17.5.1

All the devices are Sonoff Basics and have been working ok until this update.

all the devices still show in Apple Home and seem active and you can turn the switches  on and off in HomeKit however the actual devices do not change state.

the HAA manager also shows the devices as active but again when you control them nothing occurs on the actual device I.e. the lights do not turn on.

**MEPLHAA Script**

**Used device**

**HAA version**

**iOS/iPadOS version**

**Logs**
A .txt file with logs. Read [how to capture logs from your device](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs)

**Screen-shots or videos showing issue**

[image]
[image]

rickdhoy, 2024-07-30:
Sorry here is the HAA Script that was already installed and working on the previous Firmware I had not made any changes to it.

They are also Sonoff Basics R1 with toggle switches

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":
[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},
{"g":14},{"g":14,"t":0}]}]}

rickdhoy, 2024-07-31:
Ok, worked it out, for some reason the MEPLHAA Script is different and needed to be updated from the one above to the one below.

{"c":{"io":[[[12,13],2],[[0],6]],"l":13,"b":[[0,5]]},"a":[{"0":{"r":[[12]]},"1":{"r":[[12,1]]},"b":[[0]]}]}

Must have been stated somewhere but I missed it.

kpbicka, 2024-08-05:
> Ok, worked it out, for some reason the MEPLHAA Script is different and needed to be updated from the one above to the one below.
> 
> {"c":{"io":[[[12,13],2],[[0],6]],"l":13,"b":[[0,5]]},"a":[{"0":{"r":[[12]]},"1":{"r":[[12,1]]},"b":[[0]]}]}
> 
> Must have been stated somewhere but I missed it.
Nope. Your script still causes stuck ESP. I check it on v12.12.8 Merlin esp8266

Error log:
Booting rom 0.
pp_task_hdl : 3ffefcb0, prio:14, stack:51

RUS: Не работает. Тоже не могу нормально сделать конфиг на 12-ю версию