RavenSystem/esp-homekit-devices #2334: My 8266 devices seem not responding on HomeKit after a while.i have been using home assistant but before upgrading everything was ok. They seems connected to network and reply to ping  but HomeKit is offline according to hass.

freedreamer82, 2024-12-25:
**MEPLHAA Script**

**Used device**

**HAA version**
12.14.4

**iOS/iPadOS version**

**Logs**
A .txt file with logs. Read [how to capture logs from your device](https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#capture-logs)

**Screen-shots or videos showing issue**

RavenSystem, 2024-12-25:
Could you try beta version, please?

https://github.com/RavenSystem/haabeta

freedreamer82, 2024-12-26:
I have 20 devices and the issue   appears randomically.. it not so simple to update all...

aptyp222, 2024-12-27:
I have the same problem repeatedly rebooting!It doesn't help

freedreamer82, 2024-12-28:
installed latest version , i will test and let u know if is fixed.

M.

freedreamer82, 2025-01-01:
I close this, to me
Seems ok now.

Thanks

closed 2025-01-01 by freedreamer82